CadQuery/cadquery #961: Center when selecting different faces/planes

mryndzionek, 2022-01-17:
This code:
```py
import cadquery as cq

result = cq.Workplane("XY").box(1, 1, 1)
result = result.faces(">X").workplane().hole(0.5, 0.1)
result = result.faces(">Y").workplane().hole(0.5, 0.1)
```
gives this result:
[image]

Why isn't the Y-plane hole also centered? I would expect `faces` to reset center, but does not.
Also, if I change the order of selecting (Y and then X) planes it throws an exception (`Standard_ConstructionError`).

Jojain, 2022-01-17:
Because the default for the workplane origin is ProjectedOrigin

You want to use `workplane(center option="CenterOfMass"`

mryndzionek, 2022-01-17:
Hmmm, seems to be fairly standard thing, but is mentioned only once in documentation and only in context of assemblies.
Nevertheless thanks for help.

closed 2022-01-18 by mryndzionek