CadQuery/cadquery #963: Problem with `shell` and `fillet`

mryndzionek, 2022-01-18:
Below simple code:
```py
import cadquery as cq

a = cq.Workplane().box(5, 5, 5).faces(">Z").fillet(0.5)\
    .faces("<Z").shell(0.5)
    
a = a.faces("<Z").fillet(0.1)
```
gives me disappearing faces:
[image]

jmwright, 2022-01-19:
@mryndzionek There will still be weird behavior with this code, but it seems better.
```python
import cadquery as cq

a = (cq.Workplane().box(5, 5, 5)
                   .faces("<Z").shell(0.5, kind='intersection')
                   .faces(">Z").fillet(0.1))
a = a.faces("<Z").fillet(0.1)

show_object(a)
```
By default, `shell()` creates `arc` corners, so you were applying fillets to areas that had already been rounded. I rearranged your code a little bit and added `kind="intersection"` (sharp corners) to the `shell()` call, which helped a little bit. I see that @adam-urbanczyk has marked this as a kernel issue, so that's probably as good as you're going to get right now.

adam-urbanczyk, 2022-01-19:
> Looks similar on cq-editor 0.3.0dev. If it's a bug, it's a bug. Good to know I need to solve this differently. Thanks!

You can use this as a workaround:

```python
show_object(a.val().fix())
```