CadQuery/cadquery #965: [EASY QUESTION] Exporting an assembly model as .stl format

traceforward, 2022-01-19:
I can not export an assembly model which I created with cq.Assembly() with exporters.exportSTL. How can I do that. Please help.
Thanks in advance.

adam-urbanczyk, 2022-01-19:
Use `toCompound` and then export.

Jojain, 2022-01-19:
@adam-urbanczyk Since this is not the first time someone ask this, shouldn't we consider the `exportStl` to automatically make a compound of the assembly ?
I'm pretty sure it's trivial to do, and would be more transparent for users

adam-urbanczyk, 2022-01-20:
Maybe, but I really wonder what is the use of such an STL.

closed 2022-10-26 by adam-urbanczyk

jmwright, 2023-01-03:
This issue has already been addressed, but I'm going to post [this response](https://gitlab.com/kicad/libraries/kicad-packages3D-generator/-/issues/5#note_1226473401) here for archival purposes. It's a good explanation from someone in the KiCAD community of why you would want to fuse and print an assembly.