CadQuery/cadquery #970: Core dump while (trying to) sweep along unioned shapes

MoonCactus, 2022-01-26:
**Context** : I am trying to build a 2D shape, and sweep an arbitrary profile along its edges.

Not only I fail miserably at this but I even manage to get repeatable core dumps ;) For what it helps, I am a complete newcomer to cadquery but I am skilled in Openscad and many programming languages.

## To Reproduce

```python
import cadquery as cq
from cadquery import Workplane

c1s=Workplane("front").polyline([(2,0),(12,0),(11,5)]).close().offset2D(1).extrude(1)
c2s=Workplane("front").circle(6).extrude(1)
s3d = c1s.union(c2s)
s2d = s3d.faces("<Z")

brush = cq.Workplane("XZ").lineTo(0,1).threePointArc( (0.5,.5), (0,0) ).close()    # provokes a core dump
# brush = cq.Workplane("XZ").circle(1)     # results in a seriously broken object

result = brush.sweep(s2d.wires()).translate((0,0,5))
```

I am most certainly writing poor code but I failed to find how to unionize in 2D in the first place. This is the reason of such a convoluted extrusion + selection of <Z faces : **pleeease do correct me** on this matter by the way, I spent more than a day trying to get the thing done. The only option I found was to discretize the edges (or build them manually) and generate polyhedrons. This is so slow and ugly (faceted) that I would better use Openscad if I do it this way ;)

[image]

The above result is when I use a simpler "brush" to sweep the unioned shape. Otherwise with the crash I get nothing obviously.

```
$ python core_dump.py 
Segmentation fault (core dumped)
```

## Backtrace

With the "complex" brush, `cq-editor` exits with a core dump and no backtrace. With the simple circular brush there is no error, but the resulting shapes is impossible to explain in my opinion. It relly does not look like I succedded in making a rounded edge around my 2D shape ;)

`gdb` tells me this

```
Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x00007fffece560d0 in TopoDS_Builder::Add(TopoDS_Shape&, TopoDS_Shape const&) const () from /home/jeremie/miniconda3/envs/cadquery/lib/python3.9/site-packages/../../libTKBRep.so.7
```

## Environment

OS: Linux 5.4.0-92-generic #103-Ubuntu SMP Fri Nov 26 16:13:00 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Cadquery installed with miniconda

Output of `conda list` from your active Conda environment:
```
$ conda list
# packages in environment at /home/jeremie/miniconda3/envs/cadquery:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
alabaster                 0.7.12                     py_0    conda-forge
alsa-lib                  1.2.3                h516909a_0    conda-forge
aom                       3.2.0                h9c3ff4c_2    conda-forge
argh                      0.26.2          pyh9f0ad1d_1002    conda-forge
astroid                   2.9.3            py39hf3d152e_0    conda-forge
asttokens                 2.0.5              pyhd8ed1ab_0    conda-forge
atomicwrites              1.4.0              pyh9f0ad1d_0    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
autopep8                  1.5.6              pyhd8ed1ab_0    conda-forge
babel                     2.9.1              pyh44b312d_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
[... 259 lines cut ...]
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
wrapt                     1.13.3           py39h3811e60_1    conda-forge
wurlitzer                 3.0.2            py39hf3d152e_1    conda-forge
x264                      1!161.3030           h7f98852_1    conda-forge
x265                      3.5                  h4bd325d_1    conda-forge
xorg-fixesproto           5.0               h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxt                1.2.1                h7f98852_2    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yapf                      0.32.0             pyhd8ed1ab_0    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zipp                      3.7.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.11            h36c2ea0_1013    conda-forge
zstd                      1.5.2                ha95c52a_0    conda-forge
```

Jojain, 2022-01-26:
Hello, several points here :
- To add and substract 2D shapes use `Sketch`  https://cadquery.readthedocs.io/en/latest/sketch.html?highlight=sketch
- OpenSCAD users tend to use in Cadquery the same mindset they have for 3D modeling, you shouldn't the workflow is different and if you don't use it as intended you will struggle a lot more than you should. In essence you almost never need to use `ùnion` , `cut` and `intersect`, there is methods available to add and remove matter, use them !
- Finally for the sweep to work you need to place your "brush" on your path wire. (I just modified slightly your code, but it's very suboptimal, please refers to the advices above to improve it)

```python
import cadquery as cq
from cadquery import Workplane

c1s=Workplane("front").polyline([(2,0),(12,0),(11,5)]).close().offset2D(1).extrude(1)
c2s=Workplane("front").circle(6).extrude(1)
s3d = c1s.union(c2s)
s2d = s3d.faces("<Z").wires()
brush = cq.Workplane("XZ", origin=(-6.1,0,-0.5)).lineTo(0,1).threePointArc( (0.5,.5), (0,0) ).close() 

s= brush.sweep(s2d)

show_object(s)
show_object(brush, name="brush")
show_object(s2d, name="s2d")
```
[image]

[image]

MoonCactus, 2022-01-26:
Thanks Jojain, this really helps me. And right I feel like something has not yet clicked in my head.
And you are right, 10 years of OpenSCAD probably make my days harder, but I kepts on hittng its limits, hence my attempts at cadquery. It ought to be used more often by the 3DP community. The only good thing is that there is (almost only) one place to find all the doc ;)
For a while I did try to use sketches but kept on getting it wrong when I tried to sweep.
Will try again & once again thanks for the helps & recommendations

Jojain, 2022-01-26:
I wouldn't say that getting the wire of a face to make a sweep path, is the best way to do it (at least the thinking is a bit off IMO) but it can indeed be simpler for path like you show.
In this case you can retrieve the wire with : `Sketch()._faces.Wires()`

There is indeed a lot of things in the docs and it's worth reading it again and again, reading source code is also great when you want to do less obvious things. Finally you can ask questions on the [discord](https://discord.gg/n7KjvnDW) directly, it's living and you can have direct help if people are available

vincele, 2022-02-21:
@MoonCactus may have done something suboptimal or even wrong, but that is not excusing the segfault though. I think there's still something to investigate here.