CadQuery/cadquery #973: Serialize and Deserialize Shapes

SDI8, 2022-01-31:
Currently, CadQuery shapes can not be pickled. This is due to the underlying `TopoDS_Shape`.
Using `BRepTools` it _should_, however, be possible to serialize and deserialize those. 
This would open up for multiprocessing with CadQuery, among other uses.

The following example demonstrates the idea on a very simple shape.
```py
from io import BytesIO

import cadquery as cq
import OCP

box = cq.Solid.makeBox(1, 1, 1)

# Write
stream = BytesIO()
OCP.BRepTools.BRepTools.Write_s(box.wrapped, stream)
serialized = stream.getvalue()

# Read
tds_shape = OCP.TopoDS.TopoDS_Shape()
OCP.BRepTools.BRepTools.Read_s(tds_shape, BytesIO(serialized), OCP.BRep.BRep_Builder())

hopefully_box = cq.Solid.cast(tds_shape)
hopefully_box.exportStep("hopefullyBox.step")
```
My question is therefore: Is it feasible to add a custom pickling support function and register it with [copyreg](https://docs.python.org/3/library/copyreg.html), and, most importantly, would this work reliably with any shape? Are there foreseeable cases where this won't work?

Also, it looks like `BRepTools.Write_s` has some parameters currently not exposed through the `OCP` module. Maybe those might be helpful!?

Links to OCCT documentation:
* https://dev.opencascade.org/doc/refman/html/class_b_rep_tools.html#aac126917b0fd6b5319f71551c45d889d

Jojain, 2022-01-31:
You might want to take a look at this : https://github.com/CadQuery/OCP/issues/48

adam-urbanczyk, 2022-02-01:
If you need (de)serialize `cq.Shape`, you can use `cq.Shape.exportBrep` or `cq.Shape.importBrep`. If you are after full pickle support, I'd propose to implement it in OCP. You'd need need to cover `TopoDS_Shape`, `TopLoc_Location` and `gp_Vector` (maybe few more in gp). If you are interested in contributing, let's discuss further in CadQuery/OCP#48.

closed 2022-02-01 by adam-urbanczyk

SDI8, 2022-02-04:
Just for completeness sake, I have experimented with a pickle support function based on `import/exportBrep` and `copyreg`.
Take the following with a grain of salt, but it works to the extend that I can use `Shape` objects with multiprocessing.

 ```py
 def inflate_shape(data: bytes):
    return cq.Shape.importBrep(BytesIO(data))

def reduce_shape(shape: cq.Shape):
    stream = BytesIO()
    shape.exportBrep(stream)
    return inflate_shape, (stream.getvalue(),)

for c in (
    cq.Edge,
    cq.Compound,
    cq.Shell,
    cq.Face,
    cq.Solid,
    cq.Vertex,
    cq.Wire,
):
    copyreg.pickle(c, reduce_shape)
   ```