CadQuery/cadquery #976: Rectangles with offset and different heights

mryndzionek, 2022-02-02:
I can't figure out how to do something like this:
 - I have a list of tuples of points and heights, so for example [((-2, -2), 1), ((-1, 1), 2), ((0, 0), 3), ...]
 - on a sketch I would like to put a rectangle at each point with constant width, but with the height from the second element of the tuple

I managed to push the points to the sketch and then I can do a `.rect`, but obviously not with changing height. Maybe it can be dome using `offset`/`moved`?

jmwright, 2022-02-03:
`pushPoints()` accepts 3D vectors.
```python
import cadquery as cq

pts = [(-2, -2, 1), (-1, 1, 2), (0, 0, 3)]

result = cq.Workplane().pushPoints(pts).rect(1, 1)

show_object(result)
```
If you run that code in CQ-editor, you'll see it puts the rectangles in different planes how I think you want, but if you try to do something with them like extrude, CadQuery will complain that the wires are not planar. That's separate issue though, and may not matter to you.

mryndzionek, 2022-02-03:
I need the rectangles to be different height, with the height being the second element from the tuple.

adam-urbanczyk, 2022-02-03:
You can use `Sketch.each` and define a callable that returns your rectangles

mryndzionek, 2022-02-03:
Yes, I've seen `each`, but I have two problems with it:
 - when I pass a lambda to it, the lambda will receive a `Location` and nothing else. How do I return a rectangle then? Should I create a new sketch and call `rect` on it?
 - how do I pass the changing state (i.e. the varying heights) to this lambda? `Each` seems to be something like a `map`, but I think I would need something like a `fold/reduce`, or `zip`
 
Is there an example somewhere showing how to use `each`?

mryndzionek, 2022-02-03:
Okay, I managed to do to what I wanted, but not with `sketch`, just `workplane`. Even then it wasn't obvious. I used `eachpoint` with a callback that operated on `nonlocal` variables and ends with `.val().located(loc)`.
Some results:
[image]
[image]
and the final part:
[image]

adam-urbanczyk, 2022-02-03:
Your callable could be stateful. Also nothing wrong with using a loop for such a case.

mryndzionek, 2022-02-03:
Yeah, I think the crucial missing part was `.val().located(loc)`.

closed 2022-02-03 by mryndzionek

adam-urbanczyk, 2022-02-03:
That is not needed with sketch: https://github.com/CadQuery/cadquery/blob/e0b4c942ab382cac38bb462b9d878f8863a88ae7/tests/test_sketch.py#L156 .

mryndzionek, 2022-02-03:
Wasn't working for some reason. Workplane solution is okay for now. I'll use this knowledge in future projects. Thanks!