CadQuery/cadquery #977: Line or LineTo With Two Vectors

infosisio, 2022-02-04:
I am sorry for the naive question, but is there a way to pass two vectors to something like Line or LineTo in order to make a simple line segment, in a way similar to this?

result = (cq.Workplane("XY").line(1.0, 0).close().extrude(0.25))

I saw that there's makeLine that takes two vectors but it kinda gets out of the scope of simplicity as it must be combined with makeRuledSurface etc.

jmwright, 2022-02-08:
You could look at the implementation of `lineTo()` to get some inspiration on how to do this. https://github.com/CadQuery/cadquery/blob/master/cadquery/cq.py#L1601

gumyr, 2022-03-16:
As you point out, there is an `Edge` method to do this:
```python
line = cq.Edge.makeLine(cq.Vector(1, 0, 0), cq.Vector(1, 0, 0.25))
```
Why must it be combined with makeRuledSurface?  Are you trying to make a Face?

By the way, `close()` is used to create a closed wire, i.e. a shape where the starting point equals the ending point. Closing two lines to make a triangle would be a typical use of `close()`.