CadQuery/cadquery #979: Centering doesn't take direction (`direct=`) into account

Equidamoid, 2022-02-05:
The centering code always assumes no direction is specified (e.g. to center a cylinder, you translate it by `(r, r, -h/2)`).
For example, `cq.Workplane().cylinder(10, 10, (1, 0, 0))` will create a cylinder with axis going from `(0, 0, 5)` to `(0, 10, 5)`.

Since proper direction handling is a bit tricky, how about at least throwing `NotImplementedError` if centering is requested when direction is not `(0, 0, 1)`? 
This will at least save users some headache debugging suddenly shifted parts (speaking form experience).

sensille, 2023-05-14:
I'm very new to cadquery, but I think I also ran into this issue.

lorenzncode, 2024-06-01:
Duplicate of #1591.

lorenzncode, 2024-12-24:
Fixed in #1593

closed 2024-12-24 by lorenzncode