CadQuery/cadquery #984: Arbitrary-ish helical paths

altaic, 2022-02-07:
I'm trying to create helical paths around shapes other than a cylinder or cone. For example, I made this manually by splitting a helix and drawing tangent lines between the two halves:
[image]

However, I'd like to be able to do this with any closed ruled surface, or more specifically if I have a closed 2D curve on a sketch, I want to create a helix that follows that curve.

I saw that `makeHelix` works by mapping an angled line onto a surface (cylinder or cone) via u v coordinates. That sounds pretty good to me, but I'm getting lost in wrapped types and internal geometry representations. I'd like to use CQ's nice high level API if possible.

Alternatively, if it can be calculated directly with splines, I'd appreciate any pointers to papers about that. That said, I haven't gone very far down the white paper rabbit hole since I'm hoping there's an easier way.

Thanks!

adam-urbanczyk, 2022-02-07:
I don't think that it is implemented. The best would be to follow the approach of `makeHelix`. You could also try `cq.Workplane.parametricCurve` but you'd need to know the parametric equations of you helix.

altaic, 2022-02-07:
Thanks, I'll check out `parametricCurve` and figure out what's sensible from white papers.

altaic, 2022-02-08:
I got it working using `parametricCurve`, however I am looking for good alternatives. `parametricCurve` is extremely fragile and requires tweaking of `N` and `tol`, and even then it may not yield good results. In all of my tests, `N` had to be significantly lower (e.g. `N=100`) than the default (`N=400`) to get a result that wasn't completely broken.