CadQuery/cadquery #987: How to get the implemented wireframe in cadquery？

luoyu-123, 2022-02-09:
I want to get a 3D line frame of an entity in cadquery, but I don't see the corresponding column in the document. Is there such a function in cadquery?

**For example this is a picture of the entity**
[image]

**Here is the picture to get the wireframe**
[image]

################################
I saw this function on onshape. Does cadquery have a corresponding function?

Jojain, 2022-02-09:
Do you want the wire display or the wires shapes objects ?

For the display : cadquery doesn't provide any visualisation directly.
For the wires objects you can use `wires()` to get all the wires objects of your workplane

jmwright, 2022-02-09:
There are also buttons in the toolbar of CQ-editor to switch between solid and wireframe view (white and black square logos). However, that does not convert the CadQuery object to wireframe, it just shows the object that way temporarily.

luoyu-123, 2022-02-10:
Thank you so much

closed 2022-02-10 by luoyu-123