CadQuery/cadquery #988: Segmentation fault with extrusion

bragostin, 2022-02-09:
conda install : master

I have bunch of twisted wires shown in the picture below. When I try to extrude them I get a segmentation fault.
Is it expected that extrude does not work with such wires?
[image]

fedorkotov, 2022-02-09:
Are they flat self-intersecting wires or 3D wires that merely look like that from this angle?
[image]

bragostin, 2022-02-09:
Here is the top view. They are not self intersecting, they are just twisted by about 90° from one end to the other. I was just wondering if `extrude()` can be expected to work in such a case.
[image]

Jojain, 2022-02-09:
I think it is expected to work, can you share the code that gives the error, extrude (as well as other methods) has been changed recently and it may be an unexpected but

bragostin, 2022-02-09:
There is a quite long code that generates this Gyroid solid below, called `finS`.
Then  I use the following code to select the bottom faces, extract the wires (outlined in grey) and extrude:
```python
base_wires = finS.faces('<Z').wires()
base = cq.Workplane("XY")
base.ctx.pendingWires = base_wires.vals()
base = base.extrude(-10)
```
which causes the segmentation fault
[image]
.

jmwright, 2022-02-09:
Can you try extruding one of the profiles by itself instead of all of them at once?

Are the wires maybe not closed for some reason?

fedorkotov, 2022-02-09:
This conical face won't extrude either
[image]

CadQuery 2.1 (not current master) with ocp 7.5.2beta on Windows says `Cannot build face(s): wires not planar` without segfault
```python
import cadquery as cq

def truncated_cone_sector():
    return cq.Workplane("XZ")\
    .lineTo(10,0)\
    .lineTo(9,1)\
    .lineTo(0,1)\
    .close()\
    .revolve(90, (0,0,0), (0,1,0))
    
extrusion = \
    truncated_cone_sector()\
    .faces(">X")\
    .wires()\
    .toPending()\
    .extrude(2)
```

bragostin, 2022-02-09:
I extracted the first wire and it looks closed:
```python
from OCP.BRep import BRep_Tool
test = finS.faces('<Z').wires().first().vals()
print("test is closed = ", BRep_Tool.IsClosed_s(test[0].wrapped))
```
yields `test is closed =  True`
And 
```python
base_wires = finS.faces('<Z').wires().first().toPending().workplane().extrude(-2)
```
yields `StdFail_NotDone: BRep_API: command not done` (with or without `workplane()`)
So I guess it is consistent with [fedorkotov](https://github.com/fedorkotov) findings.

adam-urbanczyk, 2022-02-09:
@bragostin can you share the core dump? 

Because your wires are non-planar, I think it'd be better to make a face first (using filling) and then call extrude (there is  new overload of extrude accepting faces).

bragostin, 2022-02-09:
@adam-urbanczyk
Here it is
[core.zip](https://github.com/CadQuery/cadquery/files/8034886/core.zip)
I will try the solution using filling.
Since I want to transition from the twisted wire to a planar rectangle, maybe loft is the better solution, That is, if I manage to make it go in the right direction.

bragostin, 2022-02-10:
@adam-urbanczyk 
I tried the following to fill the wire before extruding, but I receive a `OCP.Standard.Standard_TypeMismatch: TopoDS::Edge` error.
```python
support = finS.faces('>>Z[0]').wires().first()
support = cq.Workplane('XY').interpPlate(support).extrude(-2)
```
However the following method yields the result I wanted to achieve: a unioned transition from this Gyroid twisted side subsequently unioned to a flat slab:
```python
bot_wires = finS.faces('>>Z[0]').wires().first()
ct = bot_wires.val().Center().toTuple()
xb, yb, zb = ct[0], ct[1], ct[2] 
bot_wires = bot_wires.toPending()
ex = 5.
L = 4.
W = 0.2
rect_base = (
cq.Workplane('XY').workplane().transformed(offset=cq.Vector(xb, yb, zb-ex),
rotate=cq.Vector(0, 0, 0)).rect(L, W)
)
bot_wires.ctx.pendingWires.extend(rect_base.ctx.pendingWires)
supports = bot_wires.loft()
```
[image]

bragostin, 2022-02-13:
@adam-urbanczyk  I managed to perform an extrusion following your advice, but the union is bad as can be seen in the image
[image]
The method using `loft` yields a good looking union on the other hand.
I guess this is a CAD kernel issue.