CadQuery/cadquery #990: Importing cadquery in python 3.9.7 on Windows fails

fedorkotov, 2022-02-10:
## To Reproduce

1. install latest mambaforge from here https://github.com/conda-forge/miniforge
2. activate conda `mambaforge\Scripts\activate.bat`
3. create conda environment for cadquery `mamba create -n cadquery -c cadquery cq-editor=master cadquery=master`
4. activate said environment `mamba activate cadquey`
5. run python and execute `import cadquery`

## Backtrace

```
Python 3.9.7 (default, Sep 16 2021, 16:59:28) [MSC v.1916 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import cadquery
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\KotovF\mambaforge\envs\cadquery\lib\site-packages\cadquery\__init__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "C:\Users\KotovF\mambaforge\envs\cadquery\lib\site-packages\cadquery\occ_impl\geom.py", line 5, in <module>
    from OCP.gp import (ImportError: DLL load failed while importing OCP: Не найден указанный модуль.
```

(*) `Не найден указанный модуль` = `The specified module could not be found`

## Environment

OS: Windows 8.1

Was CadQuery installed using Conda?:  yes, more precisely I used mambaforge 
Output of `conda list` from your active Conda environment: [condalist.txt](https://github.com/CadQuery/cadquery/files/8039105/condalist.txt)
Using: Python interpreter

## Additional information
Examining `procmon.exe` (One of SysInternals tools) logs I noticed that the last thing that happens when I run  `import OCP` is python unsuccessfully looking for some `libwebpmux.dll` 

See also #692

Jojain, 2022-02-10:
Can you look in your env folder if python.exe is here ? I have had recent problem where my python interpreter would be removed from my env which then lead to such import error. (There is still a pythonw.exe is the env but no python.exe)

It seems weird but that would give me similar import errors so that may be it.

fedorkotov, 2022-02-10:
`C:\Users\KotovF\mambaforge\envs\cadquery` folder which contains my environment is the first folder in PATH. 
It contains both `python.exe` and `pythonw.exe`. I checked in task manager if `python.exe` that runs when I execute `python` from cmd when the env is activated is the correct one and it is.

Jojain, 2022-02-10:
That's not a super solution but maybe try to force python 3.8 or 3.10 ( I think the later has been supported juste recently)

fedorkotov, 2022-02-10:
python 3.10 is not supported by CQ-editor
```
mamba create -n cadquery_py310 -c cadquery cq-editor=master cadquery=master python=3.10
```
```
[...]
Encountered problems while solving:
  - package cq-editor-master-py3.7 requires python_abi 3.7.* *_cp37m, but none of the providers can be installed
```

Python 3.8 environment was created successfully with command
```
mamba create -n cadquery_py38 -c cadquery cq-editor=master cadquery=master python=3.8
```

Importing cadquery failed with same error
```
Python 3.8.12 (default, Oct 12 2021, 03:01:40) [MSC v.1916 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import cadquery
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\KotovF\mambaforge\envs\cadquery_py38\lib\site-packages\cadquery\__init__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "C:\Users\KotovF\mambaforge\envs\cadquery_py38\lib\site-packages\cadquery\occ_impl\geom.py", line 5, in <module>
    from OCP.gp import (
ImportError: DLL load failed while importing OCP: Не найден указанный модуль.
```

I suspect this is caused by some obscure problem with one of dependencies as in #692
By the way I tried the recipe with downgraded `freeimage` from there and it does not work this time.

fedorkotov, 2022-02-11:
Yesterday I tried miniforge instead of mambaforge and got the same error on import.
Today I removed miniforge and installed miniconda again on the same machine. And it worked (cadquery imported successfully, CQ-editor  also works).
For reference this is the command I used for miniconda
```
conda create -n cadquery -c conda-forge -c cadquery cq-editor=master cadquery=master
```
output of conda list [list-miniconda.txt](https://github.com/CadQuery/cadquery/files/8047092/list-miniconda.txt)

I conclude that either something has changed in cadquery or one of its dependencies between my previous attempts with miniforge and now or miniforge and mambaforge are broken (at least on Windows).

adam-urbanczyk, 2022-02-11:
Could you try running ntldd on the OCP binary and share the result? You might need to install ntldd (`conda install -c conda-forge m2w64-ntldd-git`).
```
ntldd -R OCP....
ntldd --list_import OCP...
```

fedorkotov, 2022-02-12:
Here is what I did. Let me know if this is not what you meant. I've never used ntldd before
1. I recreated a broken environment
  ```
  mamba create -n cadquery_noforge -c cadquery cq-editor=master cadquery=master
  ```
2. activated this environment with `mamba activate cadquery_noforge`
3. verified that it is broken
```
Python 3.9.7 (default, Sep 16 2021, 16:59:28) [MSC v.1916 64 bit (AMD64)] :: Ana
conda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import OCP
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: DLL load failed while importing OCP: Не найден указанный модуль.
``` 
4. installed ntldd with  `mamba install -c conda-forge m2w64-ntldd-git`
5. tried executing `ntldd -R OCP`. It said `OCP: not found`
6. found the only thing that had a signature of windows executable and OCP in it's name inside environment folder - `OCP.cp39-win_amd64.pyd`
7. the following command
  ```
  ntldd -R mambaforge\envs\cadquery_noforge\Lib\site-packages\OCP.cp39-win_amd64.pyd > ntldd-R.OCP.cp39-win_amd64.pyd.txt
  ```
  produced the this file [ntldd-R.OCP.cp39-win_amd64.pyd.txt](https://github.com/CadQuery/cadquery/files/8053606/ntldd-R.OCP.cp39-win_amd64.pyd.txt)
8. the following command
   ```
  ntldd --list-imports mambaforge\envs\cadquery_noforge\Lib\site-packages\OCP.cp39-win_amd64.pyd > ntldd-list-imports.OCP.cp39-win_amd64.pyd.txt
  ```
  produced this file [ntldd-list-imports.OCP.cp39-win_amd64.pyd.txt](https://github.com/CadQuery/cadquery/files/8053607/ntldd-list-imports.OCP.cp39-win_amd64.pyd.txt)

fedorkotov, 2022-02-13:
I tried it on my Linux box (Ubuntu 20.04) today. Can not reproduce it there. mambaforge installs everything correctly even if I omit `-c conda-forge` argument in environment creation command.

On Windows machine I cleared conda's cache as described [in wiki](https://github.com/CadQuery/cadquery/wiki/Conda-tricks)  to make use of your last commit #992. And still it reproduces reliably every time
```
mamba create -n cadquery -c cadquery cq-editor=master cadquery=master
```
produces a broken environment where OCP can't be imported. And
```
mamba create -n cadquery -c conda-forge -c cadquery cq-editor=master cadquery=master 
```
installs everything properly. OCP and cadquery can be imported without issues.

Perhaps something is not right in windows version of mambaforge. Maybe it reproduces only on Windows 8.1 but not on newer versions.
I've found a workaround (just add `-c conda-forge` when using miniforge on Windows) so this can be closed if you don't have further remarks and don't need additional information from me to investigate further.