CadQuery/cadquery #991: Conda conflict between CadQuery and Fenics

bragostin, 2022-02-11:
I started a fresh conda install on Debian 11 following these steps:
```
conda create -n cadquery python=3.7
conda activate cadquery
conda install -c conda-forge -c cadquery fenics
conda install -c conda-forge -c cadquery cadquery=master
```
but then I get a never ending solve environment with a ton of conflicts and failed install at the end.
The install is flawless though with `conda install -c conda-forge -c cadquery cadquery` but then I get 2.1 version with OCP 7.4 only.
Has anybody encountered the same issue?

bragostin, 2022-02-11:
Removed the environment and started again with
```
conda create -n cadquery -c conda-forge -c cadquery fenics cadquery=master

Collecting package metadata (current_repodata.json): done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: / 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
Examining conflict for cadquery fenics:  29%|██████████████████▊                                               | 2/7 [00:05<00:13,  2.65s/\ 
```
but still no luck.

fedorkotov, 2022-02-11:
[conda is famous for glacial pace of "Examining conflict .." process](https://github.com/conda/conda/issues/10034)
Has it come to en end for you? Did it say what package requirements are in conflict? Can you post conflicts list here?

If not, check out [mamba](https://github.com/mamba-org/mamba). You can install it on top of miniconda or as part of [mambaforge](https://github.com/conda-forge/miniforge). It behaves almost identically to conda but moves almost an order of magnitude faster.
It most likely will not resolve what conda is not able to resolve but at least you will know what exactly what conflicts with what. 

One more thought. [fenics is available on PyPi](https://pypi.org/project/fenics/). Maybe you can install cadquery with conda and then fenics with pip on top of that?

bragostin, 2022-02-11:
I get 744 lines of conflicts, something must be really off.
Pipy fenics does not intall dolfin because of the C++ depndencies
I will check mamba then

bragostin, 2022-02-11:
I do have fenics installed via the Debian python distribution, so I tried adding `sys.path.append('/usr/lib/petsc/lib/python3/dist-packages')` to my python script, but then I get the error
```
    from dolfin import *
  File "/usr/lib/petsc/lib/python3/dist-packages/dolfin/__init__.py", line 34, in <module>
    from .cpp import __version__
ModuleNotFoundError: No module named 'dolfin.cpp'
```
I used to be able to install cadquery master together with fenics via conda before, something must have changed.

fedorkotov, 2022-02-11:
My conda-fu is not strong enough to provide some sure-fire recipe but until someone more knowledgeable in conda ways shows up I would extract conflicting packages and versions list from conda conflicts report and try to compare it with reсent changes in fenics and CadQuery commits to try and figure out when exactly things went sideways. Maybe it is as simple as changing a single dependency version in fenics or CadQuery.

adam-urbanczyk, 2022-02-11:
I tried with mamba (it is much faster and gives nicer error msgs in such a situation) and it seems that the hardcoded vtk dep in the CQ recipe is the root cause. I'll check if the situation is resolved and if we can remove it.

adam-urbanczyk, 2022-02-13:
I just merged #992 - after it is built pls try again (and remember to clean the conda cache: https://github.com/CadQuery/cadquery/wiki/Conda-tricks).

bragostin, 2022-02-13:
Works like a charm now, thank you!

closed 2022-02-13 by bragostin

adam-urbanczyk, 2022-02-13:
Great, are you allowed to share more details about your use case (combining CQ and fenics)?

bragostin, 2022-02-13:
I am working on it now, but the idea is to have the following workflow:
CadQuery --> Gmsh --> msh2xdmf --> Fenics to perform heat diffusion simulations.
In the example below I have Gyroïd surfaces attached to a baseplate and I want to simulate heat diffusion with heat applied below the baseplate.
The challenge here is to generate a CAD of devoid of self intersections so that Gmsh will be able to mesh it, which is not trivial with this kind of convoluted surfaces. One big challenge was to connect these curved surfaces to the flat surface at the bottom and be able to union it properly.
However as you can see below it is feasible with CadQuery.
The next step will be to feed this into Fenics.
[image]
[image]