CadQuery/cadquery #993: Missing components when exporting assembly to GLTF

jakeanq, 2022-02-12:
I'm trying to export an assembly to GLTF, as it seems to be a good way to get it into Blender, however I always get a warning during export and one piece is missing mesh data.

Reproducing using the two cones assembly example from the documentation:
```py
import cadquery as cq

cone = cq.Solid.makeCone(1, 0, 2)

assy = cq.Assembly()
assy.add(cone, name="cone0", color=cq.Color("green"))
assy.add(cone, name="cone1", color=cq.Color("blue"))
assy.constrain("cone0@faces@<Z", "cone1@faces@<Z", "Axis")

assy.solve()
assy.save("test2.gltf")
```

When run, this gives the following output:
```
RWGltf_CafWriter skipped node 'bf79109a-8bf2-11ec-860e-0090f5f2c22d_part' without triangulation data
```

The generated GLTF file looks like this when viewed in Blender:
[image]
Note that both parts show up in the scene hierarchy but only cone0 has mesh data.

I have also tried with a few other assemblies with varying numbers of parts - they all seem to produce a single warning and be missing a single part.

<details><summary>Three object version</summary>
<p>
```py
import cadquery as cq

cone = cq.Solid.makeCone(1, 0, 2)
cube = cq.Solid.makeBox(1, 1, 1)

assy = cq.Assembly()
assy.add(cone, name="cone0", color=cq.Color("green"))
assy.add(cone, name="cone1", color=cq.Color("blue"))
assy.add(cone, name="cube0", color=cq.Color("pink"))
assy.constrain("cone0@faces@<Z", "cube0@faces@<Z", "Axis")
assy.constrain("cube0@faces@>Z", "cone1@faces@<Z", "Axis")

assy.solve()
assy.save("test3.gltf")
```

Output:
```
Meshing <cadquery.occ_impl.shapes.Solid object at 0x7fe8dca77fd0> from <cadquery.assembly.Assembly object at 0x7fe8879c0910>
Meshing <cadquery.occ_impl.shapes.Solid object at 0x7fe8dca77fd0> from <cadquery.assembly.Assembly object at 0x7fe88746aec0>
Meshing <cadquery.occ_impl.shapes.Solid object at 0x7fe8dca77fd0> from <cadquery.assembly.Assembly object at 0x7fe88746afe0>
RWGltf_CafWriter skipped node '6aed5336-8bf4-11ec-936e-0090f5f2c22d_part' without triangulation data
```

For this one I dropped a print statement into the `exportGLTF` function - note that it was called 3 times:
```py
for _, el in assy.traverse():
        for s in el.shapes:
            print(f"Meshing {s} from {el}")
            s.mesh(tolerance, angularTolerance)
```

In Blender:
[image]

</p>
</details>

<details><summary>`conda list`</summary>
[... 162 lines cut ...]
tzdata                    2021e                he74cb21_0    conda-forge
utfcpp                    3.2.1                ha770c72_0    conda-forge
vtk                       9.0.1                    pypi_0    pypi
wcwidth                   0.2.5                    pypi_0    pypi
wheel                     0.37.1                   pypi_0    pypi
x264                      1!161.3030           h7f98852_1    conda-forge
x265                      3.5                  h4bd325d_1    conda-forge
xorg-fixesproto           5.0               h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxt                1.2.1                h7f98852_2    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
zlib                      1.2.11            h36c2ea0_1013    conda-forge
zstd                      1.5.2                ha95c52a_0    conda-forge
```
</p>
</details>

adam-urbanczyk, 2022-02-13:
Your top level assy is empty so I think that the warning is fine. If I use a non-empty top assy the export seems to be fine. This is either a bug of CQ or OCCT - I have too little experience with the GLTF OCCT export functionality to be able to tell.

jakeanq, 2022-02-14:
Adding something to the top-level assembly lets me work around the issue for now which is nice, thanks!

lorenzncode, 2022-09-19:
I'm testing a fix on the CQ side in #1157.

Example script:
```python
import cadquery as cq

box = cq.Workplane().box(1, 2, 3)

assy = cq.Assembly(name="toplevel")
assy.add(box, name="box1")
assy.add(box, name="box2", loc=cq.Location((4, 0, 0)))

assy.save("boxes.gltf")
assy.save("boxes.step")
```

Importing boxes.step (master) into CAD Assistant shows:
[image]

Importing boxes.gltf (master) into CAD Assistant results in error _Scene node '5' refers to non-existing mesh._

Importing boxes.step (PR1157) into CAD Assistant shows:
[image]

Importing boxes.gltf (PR1157) into CAD Assistant show:
[image]

lorenzncode, 2022-10-02:
I tested again with current master which includes #1156.  The issue is now resolved with OCCT/OCP 7.6.

Here is the cone example GLTF imported in blender: 
[image]

Importing boxes.gltf into CAD Assistant is also working now:
[image]

I'd still like to consider PR #1157 even though it is not required to fix this issue.  It would resolve an issue where an extraneous leaf part is added (at least for STEP) due to empty root assembly.

lorenzncode, 2023-02-07:
Fixed with OCP update and #1157.  Also note orientation fix in #1211.

closed 2023-02-07 by lorenzncode