CadQuery/cadquery #995: Compound shells

ReubenVandezande, 2022-02-14:
Is there a way to generate a solid from shells? I have a STEP file that imports as 3 separate shell objects and would like to combine them into a single solid.

ReubenVandezande, 2022-02-23:
I have a series of three non-watertight shells that together create a single watertight shell. I can't convert the individual shells to a solid since they aren't closed and I also can't seem to merge/fuse/combine them into a single shell. My solution was to fuse the shells in another CAD program and re-export (but that's obviously not ideal...). Is there a way to do this within cadquery?

adam-urbanczyk, 2022-02-23:
Then crate a single shell from all the faces you have (i.e. all the faces of your 3 shells) and only after that use `makeSolid`.

ReubenVandezande, 2022-03-01:
Ah, still new to this, forgot I could use the face to create a new shell. Thanks for the assistance!

closed 2022-03-01 by ReubenVandezande