CadQuery/cadquery #997: Summary of installation attempts on an M1 Mac

ornamentist, 2022-02-15:
I'm not sure an issue is the best place for these notes, if not, I'll close this and move them somewhere else.

These notes are a summary of the issues and work-arounds I've needed to install CadQuery and CQ-editor on an M1 Mac running macOS 12.x. I hope other M1 Mac users find them useful.

This is *not* a request for the developers to support M1 Macs. CadQuery is a volunteer-run open source project and I understand and accept that the developers will use their limited time as they see fit.

I _use_ a Mac for everyday tasks, but I don't _develop_ for macOS, so the following may not be accurate.

## *conda Installs

I've had no success installing CadQuery using conventional *conda installation. The underlying issue is that the OCCT libraries that CadQuery relies on are only compiled for x86_64 and *conda can't find the correct x86_64 channel to use?

## Miniforge Installs

Miniforge seemed promising so I used Homebrew to install `miniforge` and followed their instructions for using x86_64 packages on an M1 Mac:

```
$ CONDA_SUBDIR=osx-64 conda create -n [environment]
$ conda activate [environment]
$ conda env config vars set CONDA_SUBDIR=osx-64
$ conda install -c conda-forge -c cadquery cadquery=master
```

Unfortunately, this also failed to install CadQuery. I also tried the technique described here without success:

https://stackoverflow.com/questions/65415996/how-to-specify-the-architecture-or-platform-for-a-new-conda-environment-apple

## Future *conda approach

I've run out of time for trying further *conda installs of CadQuery on an M1 Mac, although I'm wondering if I start a Bash shell running as x86_64 code interpreted by Rosetta, whether *conda will pick the right channels to install x86_64 only code?

## CQ-editor / CQ-cli Installs

To avoid the issues around *conda, CadQuery and macOS on an M1 mac, I switched to using the binary builds of CQ-editor, created as part of the Github CI process for CQ-editor? These builds can be found at:

https://github.com/CadQuery/CQ-editor/releases/download/0.2/CQ-editor-MacOS.zip

Since the underlying CadQuery components are not built for an M1 Mac, compiled code in this download is strictly x86_64, which should be runable under Rosetta.

I believe compiled code on modern macOS versions needs to be signed or notarized in some way, but the files in this download are not. Once you run:

`{install dir}/CQ-editor.sh`

macOS will complain about every single compiled but unsigned object and library file. Rather than manually right-clicking and accepting every notification, I used this Bash script (which I found somewhere, but didn't record where):

```
macallow () {
  if [[ $# -gt 0 ]]; then
    for f in $@; do
      sudo xattr -d -r com.apple.quarantine $(realpath $f)
    done
  else
    echo syntax: macallow [filename or wildcard]
  fi
}
```

And then:

```
$ source allow.sh
$ macallow $(find {install-dir} -name '*.so')
$ macallow $(find {install-dir} -name '*.dylib')
$ macallow $(find {install-dir} -name 'Qt*')
```

Which made all the unsigned files runable under Rosetta. A similar approach also worked with the downloads of `CQ-cli`.

ornamentist, 2022-02-15:
Okay, this worked:

```
# install miniforge as per instructions
# start a new shell or source the lines added to shell rc file
$ CONDA_SUBDIR=osx-64 conda create -n rosetta python
$ conda env config vars set CONDA_SUBDIR=osx-64
$ conda deactivate 
$ conda activate rosetta
$ python -c "import platform;print(platform.machine())"
$ conda install -c conda-forge -c cadquery cadquery=master
$ python -c "import cadquery as CQ"
```

Reference: https://github.com/conda-forge/miniforge/issues/165

michaelgale, 2022-02-15:
Thanks for making these helpful notes!  It will be good to put one of my M1 Mac mini's to use as a CQ rendering resource.

adam-urbanczyk, 2023-01-25:
conda-forge provides osx-arm64 builds of ocp

closed 2023-01-25 by adam-urbanczyk

yehoshuapw, 2023-08-02:
in case this helps someone, I've managed to install on aarch64 (arm64) ubuntu. I've had the prev version in a similar manner, but didn't write down what worked, and it was too much of a mess to duplicate.

I've now upgraded, and hopefully this should work for others too:

* install vtk-9.1 (this may have needed installing casadi & chemfiles, but I didn't need to touch that again)
  * due to some name clash, I've renamed `AllValues` to `VtkAllValues` (added patch from tag v9.1.0)
* get sources
```
git clone --recurse-submodules https://github.com/Open-Cascade-SAS/OCCT.git
git clone --recurse-submodules https://github.com/CadQuery/OCP.git
```
 
 * build and install OCCT
```
cd ./OCCT
git checkout V7_7_1
mkdir build
cd build
sudo touch /usr/local/bin/custom.sh
sudo chown `whoami` /usr/local/bin/custom.sh # cmake tried to use it
cmake -DINSTALL_DIR=/usr/local .. -DUSE_VTK=ON -DBUILD_PATCH=/home/jupyter/src/OCP/opencascade -D3RDPARTY_VTK_INCLUDE_DIR=/usr/local/include/vtk-9.1/
make -j4
sudo make install
```

* build and install OCP
```
cd ./OCP
# install pywrap
cd ./pywrap
pip install . --user
cd ..

pywrap -l /usr/lib/llvm-14/lib/libclang.so -i /usr/local/include -i /usr/local/include/vtk-9.1/ all ocp.toml

pip install --user 'pybind11>=2.9.2'
mkdir ./build
cd build
# linker issue using gcc, so using clang.
export CC=/usr/bin/clang
export CXX=/usr/bin/clang++
cmake -Dpybind11_DIR=/home/jupyter/.local/lib/python3.10/site-packages/pybind11/share/cmake/pybind11 -DPYTHON_EXECUTABLE=/usr/bin/python3.10 -DOPENCASCADE_INCLUDE_DIR=/usr/local/include/opencascade -DOPENCASCADE_LIBRARIES=/usr/local/lib/libTKernel.so ..
make -j4
sudo make install
```

* install cadquery
```
pip install -U 'cadquery>=2.3.0'
```

something here is still missing (the cadquery-ocp python wheel.), I'll try to re-do that part again and write it here too.