CadQuery/cadquery #998: occt 7.6.0 support isn't ready yet

greyltc, 2022-02-15:
I'll open this issue just to track 7.6.0 readiness, I understand it's in the works, but just not merged yet.

Yesterday I build built OCP ([34e223c](https://github.com/CadQuery/OCP/commit/34e223c89c2eae0dbf9695a0f76eb3fc28f6f939)) with OCCT 7.6.0 and that seems to go just fine, but when I try to use that with the latest in master (4108c7d) from this repo, the build goes okay, but the tests fail like this:
```
==> Starting check()...                                                                            
======================================= test session starts =======================================
platform linux -- Python 3.10.2, pytest-6.2.5, py-1.11.0, pluggy-0.13.1                            
rootdir: ~/git/python-cadquery-git/src/cadquery
plugins: hypothesis-6.17.0, cov-2.12.1, anyio-3.5.0, forked-1.4.0
collected 0 items / 13 errors                                                                     

============================================= ERRORS ==============================================
_____________________________ ERROR collecting tests/test_assembly.py _____________________________
ImportError while importing test module '~/git/python-cadquery-git/src/cadquery/tests/test
_assembly.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
/usr/lib/python3.10/importlib/__init__.py:126: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
tests/__init__.py:1: in <module>
    from cadquery import *
cadquery/__init__.py:3: in <module>
    from .occ_impl.shapes import (
cadquery/occ_impl/shapes.py:59: in <module>
    from OCP.BRepAdaptor import (
E   ImportError: cannot import name 'BRepAdaptor_HCurve' from 'OCP.BRepAdaptor' (unknown location)
```

altaic, 2022-03-06:
FYI, from the OCCT git repo commit where `BRepAdaptor_HCurve` is removed:

> [0028966](https://git.dev.opencascade.org/gitweb/?p=occt.git;a=object;h=0028966): Coding Rules - remove Adaptor2d_HCurve2d, Adaptor3d_HCurve and Adaptor3d_HSurface classes
> 
> Adaptor2d_Curve2d, Adaptor3d_Curve and Adaptor3d_Surface now inherit Standard_Transient.
> Interfaces Adaptor2d_HCurve2d, Adaptor3d_HCurve, Adaptor3d_HSurface and their subclasses
> are now aliases to Adaptor2d_Curve2d, Adaptor3d_Curve and Adaptor3d_Surface.
> Removed numerous unsafe reinterpret casts.
> 
> Generic classes Adaptor3d_GenHCurve, Adaptor3d_GenHSurface, Adaptor2d_GenHCurve2d have been removed.
> Several redundant .lxx files have been merged into .hxx.
> 
> Removed obsolete adaptor classes with H suffix.

altaic, 2022-03-06:
More info here: https://github.com/CadQuery/OCP/issues/68#issue-1065328768

Also, note that OCCT 7.6.1 is out with “critical” bug fixes.

darrellenns, 2022-04-01:
The occt7.6 branch hasn't had a commit in months. Does anyone know what the state of it is? Are there issues with moving to 7.6, or do we just need a pull request to get it into master?

whophil, 2022-04-12:
Is this dependent on an OCP 7.6 release? Upstream issue: https://github.com/CadQuery/OCP/issues/85

kiufta, 2022-09-07:
@greyltc What's the current status? Just built aginst OCCT 7.6.3:

`ImportError: cannot import name 'BRepAdaptor_HCurve' from 'OCP.BRepAdaptor' (unknown location)`

closed 2022-10-01 by adam-urbanczyk