CadQuery/cadquery #999: Unexpected mode='a' (fuse) behavior in Sketch

fedorkotov, 2022-02-15:
The following script
```python
import cadquery as cq

s = (
     cq.Sketch()
    .trapezoid(2,3,80)
    .trapezoid(2,3,100, mode='a')
    .vertices()
    .fillet(.2)
)

x = (cq.Workplane("XY")
     .placeSketch(s)
     .extrude(0.5))
```
produces this
[image]

I thought the result would be the same as for sketch with the first trapezoid commented
[image]
I expected the first trapezoid to disappear because the second one fully encompasses it and was added in fuse mode (`mode='a'`).
Is this an expected behavior?

adam-urbanczyk, 2022-02-15:
Yes it is expected, no implicit cleans are performed (i.e. calls to `ShapeUpgrade_UnifySameDomain`). This does what you want:
```python
import cadquery as cq

s = (
     cq.Sketch()
    .trapezoid(2,3,80)
    .trapezoid(2,3,100, mode='a')
    .clean()
    .vertices()
    .fillet(.2)
)

show_object(s)
```

fedorkotov, 2022-02-15:
Oh, I see  now what `clean()` does. Thank you. 

Docstring for `Shape.clean(..)` says "Experimental clean using ShapeUpgrade". It is not easy to guess from that what is it for if one is not familiar with OpenCascade API (like me).
Maybe a [full description](https://dev.opencascade.org/doc/refman/html/class_shape_upgrade___unify_same_domain.html#details) from OpenCascade docs can be copied in there? 
Or is the docstring deliberately vague because the method will be changed to do something slightly different in near future?

adam-urbanczyk, 2022-02-15:
This is the docstring you should be looking at:
https://github.com/CadQuery/cadquery/blob/23de5cc3e4667c7875d2dd504564045915b4e866/cadquery/sketch.py#L547

Minimalist, but to the point.

fedorkotov, 2022-02-15:
Sorry. My bad. I was looking at the wrong `clean(..)` in https://cadquery.readthedocs.io/en/latest/classreference.html.

closed 2022-02-15 by fedorkotov