CadQuery/cadquery #1002: Step -> Stl convert script does not output file

kfina91, 2022-02-17:
I have a really simple script:

```
// read command line arguments to args...
model = cq.importers.importStep(args.input)
cq.exporters.export(model, args.output)
```

This works for most models, but for some models the code runs, but does not generate the file and does not throw any exceptions. It just silently fails to write the file.
Unfortuantely I can not share the .step file in question due to licensing.

If I import and export the .step in question in FreeCAD the code runs through and from the runtime, I deduce that already the importing step is not working for the original file.

Can I obtain any info on why it is failing, or can I detect failing files beforehand with some logic?

jmwright, 2022-02-18:
Please note, that without a STEP file that reproduces the problem, there's no way to know if the above will solve your issue.

kfina91, 2022-02-18:
Thank you for your help!

I'll close this, since this is a OCCT issue rather than a cadquery problem.

closed 2022-02-18 by kfina91