CadQuery/cadquery #1003: Mesh example

paireks, 2022-02-18:
Hi! :)

I've tried so hard to figure out if there is any way to create triangulated meshes using cadquery, and couldn't found simple answer or a working example. I've found e.g. this: https://groups.google.com/g/cadquery/c/C1mMNHhBo8U/m/fdAbqQdvAwAJ, but it's not working.

Is there any example how to create such shapes even if there is no built-in method for that?

fedorkotov, 2022-02-18:
By "triangulated meshes" you mean surface meshes for rendering or 3D printing? 
Then use this section of documentation as a starting point  [triangulated meshes](https://cadquery.readthedocs.io/en/latest/importexport.html?highlight=stl#exporting-stl).

Or do you mean surface and volume meshes for use in some CAE (CFD, stress analysis, etc)?
CadQuery and OpenCascade  can not do this. More precisely they can be part of the solution. You can build your geometric model in CadQuery but then you should export it model into something like STEP and use a specialized programs to mesh it (for example GMSH https://gmsh.info/)

paireks, 2022-02-18:
By triangulated meshes I mean meshes for rendering. I see that there are some sort of importers or exporters for STL, but is there a way to build such mesh without it? Or should I create an STL file and then import it to cadquery? :)

adam-urbanczyk, 2022-02-18:
Is this what you mean: https://github.com/CadQuery/cadquery/blob/9e9b452e66f7925f7c15689f3e07da47b13f2454/cadquery/occ_impl/shapes.py#L1137 ?

adam-urbanczyk, 2022-02-18:
CQ is B-rep based (OCCT is the CAD kernel behind it) and does not support meshes for modeling. Are you sure that you are looking in the right place? What is your end goal?

adam-urbanczyk, 2022-02-19:
If you are referring to https://github.com/bernhard-42/jupyter-cadquery, it uses pythreejs. CQ has some simple rendering based on vtk.js. Better ask questions on the forums of those projects.

paireks, 2022-03-13:
@adam-urbanczyk Ok, in the end I actually use cadquery, but using tessellate method to create some meshes, as it is sometimes much easier to model some BREPs and convert them to meshes. Here is notebook example, maybe someone will find it useful ;)
https://github.com/paireks/dotbimpy/blob/master/dotbimpy/other/WallsWithBeams.ipynb