CadQuery/cadquery #1004: Question over code location - Importing conda cadquery from virtual environment.

grbd, 2022-02-18:
Hi,
I recently put together a pull request for a web installer for installing cadquery / usefull bits
as part of a self installer exe / sh

  * https://github.com/CadQuery/cadquery/pull/1000

One of the effects of this installer is that it places a cadquery conda instance on a path we can look for.
So I then came up with a small module of code that searches for it, then adds it to the sys.path etc
The idea being you can then easily access cadquery from a virtual python setup that's not conda related.

This seems to work quite well and avoids the need for python wheels
which have issues with file size restrictions and building C++ projects.
(I've been through the pain of building C++ projects myself while building freecad, so I appreciate why conda is required)

All it requires is a single import statement above the import cadquery statement
```python
import cadquery_external
import cadquery
```

I was going to do this as a pull request
However I can't place this inside of the main cadquery repo as __init__.py gets called first
and that then tries to pull in OCP which it won't be able to find yet

So any thoughts on the best place to put this?
Maybe a seperate python package that can be installed via pip (although it seems a bit small for that)

**cadquery_external.py**
```python
import sys
import os
from os import environ, path

# This allows for using cadquery from an external non conda python installation
# Such as a python virtual environment
# Run via "import cadquery_external"
# Or "from cadquery_external import import_cadquery_conda"
#    "import_cadquery_conda(path_to_cadquery_install)"

def find_cadquery_conda_dir():
    """Search for installed cadquery conda instance"""

    # Search the Home directory
    cadquery_home = path.join(path.expanduser("~"), "cadquery")
    if path.isdir(cadquery_home):
        return cadquery_home

    # Search the ProgramData directory for system wide installs
    if sys.platform == "win32":
        cadquery_progfiles = path.join(environ["ALLUSERSPROFILE"], "cadquery")
        if path.isdir(cadquery_progfiles):
            return cadquery_progfiles

    return None

def import_cadquery_conda(condapath=None):
    """Attempt to locate and import cadquery from an external conda install"""
    if condapath is None:
        condapath = find_cadquery_conda_dir()
    if condapath is not None:
        site_dir = path.join(condapath, "lib/site-packages")
        sys.path.append(site_dir)
        bin_dir = path.join(condapath, "Library/bin")
        os.add_dll_directory(bin_dir)

import_cadquery_conda()
```

adam-urbanczyk, 2022-02-18:
I don't follow at all.  The user will be able to configure his or her own env better and without making assumptions on the install location.

grbd, 2022-02-19:
I think I'll try and add it as an example to the contrib repo.

The general gist of what I'm aiming for is making things as simple as possible for using cadquery outside of conda.
So under windows for example

  * double click an exe, then click next next next etc
  * install ends up in C:\Users\username\cadquery or C:\ProgramData\cadquery based on which option you pick on the installer
  * Or under Linux it defaults to ~/cadquery/
  * then if you want to use it outside of the python bundled with conda, you just copy / paste a single file and add a one liner to the python file above "import cadquery"

closed 2022-02-19 by grbd