CadQuery/cadquery #1008: CutThruAll breaks when shapes overlap.

TeddyMori, 2022-02-21:
When trying to use `cutThruAll` in situations where the 2d shapes overlap causes an error. I would expect the cut shape to be the union of the shapes used. My current workaround is making the shapes I would normally want to cut  through manually, union-ing them, then subtracting them from the base shape, but that makes the code so much dirtier. 

## To Reproduce

Seems like any situation where shapes are overlapping before `cutThruAll()` is called causes an exception. Here, any value >= 10 for the circle causes the exception.

```python
coords = [(-10,0), (10, 0)]
shape = cq.Workplane('XY').box(100, 100, 10)
shape.pushPoints(coords).circle(10, 0).cutThruAll()
```
Looking at the result of `shape1 = cq.Workplane('XY').pushPoints(coords).circle(11).extrude(10)` seems to give a hint as to what is happening? Seems like the ">Z" and "<Z" faces are gone?

[image]

## Backtrace

```
---------------------------------------------------------------------------
Standard_ConstructionError                Traceback (most recent call last)
/tmp/ipykernel_16486/1333192834.py in <module>
      1 coords = [(0,0), (20, 0)]
      2 shape = cq.Workplane('XY').box(200, 200, 10)
----> 3 shape.pushPoints(coords).circle(10, 0).cutThruAll()

/mnt/chromeos/MyFiles/repos/cadquery/cadquery/cq.py in cutThruAll(self, clean, taper)
   3484         solidRef = self.findSolid()
   3485 
-> 3486         s = solidRef.dprism(
   3487             None, self._getFaces(), thruAll=True, additive=False, taper=-taper
   3488         )

~/anaconda3/envs/cadquery/lib/python3.9/site-packages/multimethod/__init__.py in __call__(self, *args, **kwargs)
    301         func = self[tuple(func(arg) for func, arg in zip(self.type_checkers, args))]
    302         try:
--> 303             return func(*args, **kwargs)
    304         except TypeError as ex:
    305             raise DispatchError(f"Function {func.__code__}") from ex

/mnt/chromeos/MyFiles/repos/cadquery/cadquery/occ_impl/shapes.py in dprism(self, basis, faces, depth, taper, upToFace, thruAll, additive)
   2590                 feat.Perform(upToFace.wrapped)
   2591             elif thruAll or depth is None:
-> 2592                 feat.PerformThruAll()
   2593             else:
   2594                 feat.Perform(depth)

Standard_ConstructionError: gp_Dir2d() - input vector has zero norm
```

## Environment

OS: Ubuntu

Was CadQuery installed using Conda?: 

It's a development install (`python setup.py develop`). I'm pretty sure I had the problem on 2.1 from conda as well though.  

Last commit from git log:
```
commit 6c8502fbf366d72f3dd004c43c96b16d5b322c2c (HEAD -> master, origin/master, origin/HEAD)
Merge: 9ef08d6 9097fe0
Author: Jeremy Wright <wrightjmf@gmail.com>
Date:   Thu Jan 6 13:12:56 2022 -0500

    Merge pull request #946 from lorenzncode/issue-944
    
    Face.makeFromWires - check that wires are closed
```

Output of `conda list` from your active Conda environment:
[... 242 lines cut ...]
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
webcolors                 1.11.1                   pypi_0    pypi
webencodings              0.5.1                    pypi_0    pypi
websocket-client          1.2.3              pyhd8ed1ab_0    conda-forge
websockets                10.1                     pypi_0    pypi
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
widgetsnbextension        3.5.2                    pypi_0    pypi
x264                      1!161.3030           h7f98852_1    conda-forge
x265                      3.5                  h4bd325d_1    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxt                1.2.1                h7f98852_2    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zipp                      3.6.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.11            h36c2ea0_1013    conda-forge
zstd                      1.5.1                ha95c52a_0    conda-forge
```

Using: Jupyter

adam-urbanczyk, 2022-02-21:
You'll need to make sure that your wires/edges do not intersect. If you cannot do it, use `Sketch` to fuse them first.

TeddyMori, 2022-02-21:
Is it reasonable to at least add a more useful error?

adam-urbanczyk, 2022-02-21:
Not really feasible with OCP/OCCT and we are not going to use union/intersection check to keep things fast. As I mentioned above - use `Sketch` if you want to support this use case.

In general B-rep is less robust - if you don't really need it you might enjoy CSG more.

TeddyMori, 2022-02-25:
Alright, bummer that I have to use sketch but I guess this can be closed

closed 2022-02-25 by TeddyMori