CadQuery/cadquery #1010: Fuse fails to clean a face that is based on a somewhat tangled wire

voneiden, 2022-02-21:
1) Similar as in CQ SVG export, the wire is created from projected edges using cq.Wire.combine. The result has 7 edges.
2) The wire edge order is shown in screenshot, it forms a sort of number "8" loop:

[image]

3) The wire is then converted to face using cq.Face.makeFromWires - the face looks OK, but the underlying Wire is still tangled
4) Further attempts to fuse, combine, clean, etc. fail to clean up the underlying wire.
5) Exporting face to brep shows the 7 edges still there in the brep file, however, **importBrep appears to clean the wire (!)**
6) The wire brep seems to be imported with the original 7 edges intact (seems correct).

Attached are both breps for the face and the wire.
[breps.tar.gz](https://github.com/CadQuery/cadquery/files/8111981/breps.tar.gz)

voneiden, 2022-02-21:
I also tried to fuse these two faces (illustrated in wireframe)
[image]

and the result was a face like this (illustrated in wireframe)
[image]

voneiden, 2022-02-21:
Tested the imports like this

```
import cadquery as cq
import logging

logger = logging.getLogger(__name__)

face = cq.Face.importBrep('projected_face.brep')
wire = face.outerWire()
wire2 = cq.Face.importBrep('projected_wire.brep')
logger.info(face.Area())
show_object(face, 'face') # Face is correct size
show_object(wire, 'wire') # Clean wire with 4 edges
show_object(wire2, 'wire2') # Tangled wire, OK as expected
```

jmwright, 2022-02-28:
I wonder if [sortWiresByBuildOrder](https://github.com/CadQuery/cadquery/blob/7847224ff9fab1dfe6a5fced0167dd35e9c1d2d8/cadquery/occ_impl/shapes.py#L3410) would help here.

voneiden, 2022-03-06:
In this case there's only one wire, sortWiresByBuildOrder expects two or more. Under the hood it uses Face.makeFromWires, which does not resolve the issue (step 3 in the original description).

edgesToWires does not change anything either (uses ShapeAnalysis_FreeBounds.ConnectEdgesToWires).