CadQuery/cadquery #1021: CutBlind() and hole() return face instead of solid

bragostin, 2022-03-04:
The following code
```python
EV = cq.Workplane("YZ").box(100,50,10).faces("<X").polygon(12, 5).cutBlind(-4)
```
returns a face with the cut polygon instead of the solid as show in the image below. The same happens with
```python
EV = cq.Workplane("YZ").box(100,50,10).faces("<X").hole(5/2, 4)
```
It used to return the solid with the hole before.
[image]

Jojain, 2022-03-04:
You need to create a workplane after your face selection if you want these operations to work on the context solid

bragostin, 2022-03-04:
with
```python
EV = cq.Workplane("YZ").box(100,50,10).faces("<X").workplane().hole(5/2, 4)
```
the issue remains. Does it work with you?

adam-urbanczyk, 2022-03-05:
AFAIR it should not be needed to use `workplane`. Maybe there is sometyhing wrong with `findSolid`? https://github.com/CadQuery/cadquery/blob/5147dd83fd2d7b8cd3715964c367ebac1b5278a5/cadquery/cq.py#L784

bragostin, 2022-03-05:
My bad, I had added `Face` to `self._findType` to try something quickly and forgot to remove it after...
Its works as expected now, but it sill needs `workplane()` after the face selection, otherwise I jut get the solid without hole.

closed 2022-03-05 by bragostin