CadQuery/cadquery #1024: exportSVG on workplane with Sketch

truell20, 2022-03-05:
Can sketches be exported to SVGs? I'm running this program:

```
import cadquery as cq

result = cq.Workplane("front").sketch().circle(1).parent
result.exportSvg('test.svg')
```

This is the error I get:

```Traceback (most recent call last):
  File "/Users/mntruell/test_cq.py", line 5, in <module>
    result.exportSvg('test.svg')
  File "/usr/local/Caskroom/miniforge/base/envs/myenv/lib/python3.10/site-packages/cadquery/cq.py", line 1079, in exportSvg
    exportSVG(self, fileName)
  File "/usr/local/Caskroom/miniforge/base/envs/myenv/lib/python3.10/site-packages/cadquery/occ_impl/exporters/svg.py", line 290, in exportSVG
    svg = getSVG(shape.val(), opts)
  File "/usr/local/Caskroom/miniforge/base/envs/myenv/lib/python3.10/site-packages/cadquery/occ_impl/exporters/svg.py", line 167, in getSVG
    uom = guessUnitOfMeasure(shape)
  File "/usr/local/Caskroom/miniforge/base/envs/myenv/lib/python3.10/site-packages/cadquery/occ_impl/exporters/svg.py", line 66, in guessUnitOfMeasure
    bb = BoundBox._fromTopoDS(shape.wrapped)
AttributeError: 'Sketch' object has no attribute 'wrapped'
```

lorenzncode, 2022-03-05:
No, currently I don't think you can directly export the `Sketch` to SVG.  You could try exporting the `Face`.
```
import cadquery as cq

s = cq.Sketch().circle(1)
cq.exporters.export(s._faces, "out.svg", opt={"projectionDir": (0, 0, 1)})
```

truell20, 2022-03-10:
That worked! Thanks.

roipoussiere, 2022-03-23:
It works in dxf pretty well. So this works:

    cq.exporters.export(cq.Workplane().placeSketch(sketch), op.join(output_path, 'sketch.dxf'))

But not this:

    cq.exporters.export(cq.Workplane().placeSketch(sketch), op.join(output_path, 'sketch.svg'))

I think that the ability to export sketches in svg is a common use case and it makes sense to simplify the process.
Could this be reopened?

lorenzncode, 2022-10-05:
Now with the Sketch `val` method:

```py
import cadquery as cq

s = cq.Sketch().circle(1)
cq.exporters.export(s.faces().val(), "out.svg", opt={"projectionDir": (0, 0, 1)})
```

roipoussiere, 2022-10-06:
Perfect! I suppose this can be closed now?

Also, maybe this solution could be mentioned in the export chapter of the documentation.