CadQuery/cadquery #1026: Scaling a STEP file in one dimension

hexad, 2022-03-10:
I want to perform a scaling operation to a STEP file that i imported.

The problem is, i want to be scaled only on one dimension. Searching the docs (https://cadquery.readthedocs.io/en/latest/classreference.html), i found the scale() function.

However, looking at the source code, 
```
    def scale(self, factor: float) -> "Shape":
        """
        Scales this shape through a transformation.
        """

        T = gp_Trsf()
        T.SetScale(gp_Pnt(), factor)

        return self._apply_transform(T)
```
it does not accept axis information.

So, is it possible to scale an item only on one axis?

hexad, 2022-03-11:
Maybe i can do this using a scaling matrix?

```
t = cq.Matrix([
        [x, 0, 0, 0],
        [0, y, 0, 0],
        [0, 0, z, 0],
        [0, 0, 0, 1]
    ])
```

Any help with how this can be incorporated?

jmwright, 2022-03-12:
There is some code in [this issue](https://github.com/CadQuery/cadquery/issues/209) that might help you.

hexad, 2022-03-14:
Thanks. I tried the code there, but CadQuery has trouble importing the libraries. I get `ModuleNotFoundError: No module name 'OCC'`.
I am using the CQ-Editor, standalone application, so how come OCC is not recognized?

hexad, 2022-03-15:
Thank you! How exactly do i select the axis and where do i change the scaling factor?

hexad, 2022-03-15:
The xs is the scaling factor and its location on the identity matrix changes the scaling axis?

fedorkotov, 2022-03-15:
> The xs is the scaling factor and its location on the identity matrix changes the scaling axis?
See this https://en.wikipedia.org/wiki/Scaling_(geometry) and this https://en.wikipedia.org/wiki/Affine_transformation

hexad, 2022-03-15:
I tried this:

```
imported_STEP = cq.importers.importStep('./CQ/no_lvl_curved.STEP')
xs = 4
xform = gp_GTrsf()
xform.SetVectorialPart(gp_Mat(
xs, 0, 0,
0, 1, 0,
0, 0, 1,
))

brep = BRepBuilderAPI_GTransform(imported_STEP.wrapped, xform, False)
brep.Build()
fshape = brep.Shape()
R1 = cq.Shape.cast(fshape)
show_object(R1)
```

but i get this error:
`AttributeError: 'Workplane' object has no attribute 'warped'`

jmwright, 2022-03-15:
Are you sure that the error says `warped`? It should say `wrapped`. I think you want `imported_STEP.wrapped` to become `imported_STEP.val().wrapped`

hexad, 2022-03-15:
You were right, it was `wrapped`. Thank you very much! This worked. As far as i can tell, CQ can run all the functions of the OpenCascade library? Do you have any good documentation reference for them? Once again, thanks a lot!

jmwright, 2022-03-16:
For the OCP layer, I rely on the OpenCASCADE documentation. https://dev.opencascade.org/doc/overview/html/