CadQuery/cadquery #1028: Objects selection on sketches

Jojain, 2022-03-13:
It is not possible to selects objects if the object on the stack is a sketch

```python
s = Sketch().rect(10,5)
b = Workplane().placeSketch(s).wires()
print(b.ctx.pendingWires) # return []
```
I needed this because in order to loft several sketchs I need to push their wires onto the context otherwise they arent usable anymore after workplane offsettings.

I think this should also be an issue, I feel it should be possible to do : 

``` python
s = Sketch().rect(10,5)
b = Workplane().placeSketch(s).workplane(10).placeSketch(s).loft()
```

gumyr, 2022-04-07:
As a workaround, this works for me:
```python
        cup = Workplane(
            cup_sketch._faces.Faces()[0].rotate(Vector(), Vector(0, 1, 0), -90)
        ).wires()
```

Lofting sketches is shown in the docs:
```python
from cadquery import Workplane, Sketch, Vector, Location

s1 = (
     Sketch()
     .trapezoid(3,1,110)
     .vertices()
     .fillet(0.2)
     )

s2 = (
     Sketch()
     .rect(2,1)
     .vertices()
     .fillet(0.2)
     )

result = (
    Workplane()
    .placeSketch(s1, s2.moved(Location(Vector(0, 0, 3))))
    .loft()
    )
```

jdegenstein, 2022-06-08:
Building on Jojain's issue above, I have also noticed that loft() does not seem to accept a mixture of Sketches and wires. According to gumyr, I believe this may result from the sketch not being cast into a wire?

```py
base = 23
crad = 6

def base_sketch_wire(self,scale=1): #hacky stuff to convert sketch to a wire
    w = self.sketch().rect(scale*base,scale*base).vertices().fillet(scale*crad)._faces.Faces()[0].outerWire()
    return self.eachpoint(lambda loc: w.moved(loc), True)

cq.Workplane.base_sketch_wire = base_sketch_wire

s1 = (cq.Sketch()
  .rect(base,base)
  .vertices()
  .fillet(crad)
  )

s2 = (cq.Sketch()
  .rect(.75*base,.75*base)
  .vertices()
  .fillet(crad)
  )

#taken from docs, and works correctly
f1 = (cq.Workplane()
  .placeSketch(s1,s2.moved(Location(Vector(0, 0, 7))))
  .loft()
  )

#does NOT work
f2 = (cq.Workplane()
  .placeSketch(s1)
  .workplane()
  .transformed(offset=cq.Vector(0,0,5))
  .placeSketch(s2)
  .loft()
  )

#works with hacky function to convert sketch to wire
f3 = (cq.Workplane()
  .base_sketch_wire(1)
  .workplane()
  .transformed(offset=cq.Vector(0,0,5))
  .base_sketch_wire(.75)
  .loft()
  )

#does not work
f4 = (cq.Workplane()
  .placeSketch(s1)
  .workplane()
  .transformed(offset=cq.Vector(0,0,5))
  .base_sketch_wire(.75)
  .loft()
  )
```