CadQuery/cadquery #1035: How to use the Export GLTF feature

kfina91, 2022-03-28:
I have seen that you merged the GLTF export feature branch.

My question is: How do you use it?
I installed cadquery via "conda install -c cadquery cadquery cadquery=master" 
(I got this from here: https://anaconda.org/cadquery/cadquery, since the command in the README.md still refers to the old repo by dcowden.)

If I open python then and write:
```
import cadquery as cq
dir(cq.exporters.ExportTypes) # prints ['AMF', 'DXF', 'STEP', 'STL', 'SVG', 'TJS', 'VRML', 'VTP', <private variables>]
a = new cq.Assembly()
dir(a) # doesn't print any export functions
dir(a.toCompound()) # prints [..., 'exportBrep', 'exportStep', 'exportStl', ...]
```

I also tried just using: `cq.exporters.export(model, "filename.glb", exportType="GLTF")`. But it failed as well with ValueError("Unknown export type").

jmwright, 2022-03-28:
The following works for me and is pulled from CadQuery's test suite.
```python
import cadquery as cq
b1 = cq.Workplane().box(1, 1, 1).faces("<Z").tag("top_face").end()
b2 = cq.Workplane().box(1, 1, 1).faces("<Z").tag("bottom_face").end()
b3 = cq.Workplane().pushPoints([(-2, 0), (2, 0)]).tag("pts").sphere(1).tag("boxes")

assy = cq.Assembly(b1, loc=cq.Location(cq.Vector(0, 0, 0)), name="TOP")
assy2 = cq.Assembly(b2, loc=cq.Location(cq.Vector(0, 4, 0)), name="SECOND")
assy2.add(b3, loc=cq.Location(cq.Vector(0, 4, 0)), name="BOTTOM")

assy.add(assy2, color=cq.Color("green"))
assy.save("/home/jwright/Downloads/boxes.glb", "GLTF")
```

kfina91, 2022-03-29:
Thank you.

One note for other people checking this post: It's important on how you add a part to your assembly for export:
```
# does not work:
part = ...
assy = cq.Assembly()
assy.add(part)
assy.save("filename.glb", "GLTF")

# does work
part = ...
assy = cq.Assembly(part)
assy.save("filename.glb","GLTF")
```

closed 2022-03-29 by kfina91