CadQuery/cadquery #1036: Black version 19.10b0 check failing

lorenzncode, 2022-03-30:
See black --check fail in #1018.

Black issue is psf/black#2964

Perhaps consider updating pin to more recent version of Black?

jmwright, 2022-03-30:
Any idea why this error showed up now and not on any of the latest PRs? Here is [an issue](https://github.com/CadQuery/cadquery/pull/644) that gives a one line explanation of why the black version was pinned. According to that, I think we could update the black version as long as the formatting behavior stays the same.

I also just noticed while looking at the log that OCP is pinned to version 7.5.1 in the AppVeyor config, so I need to fix that.

lorenzncode, 2022-03-30:
> Any idea why this error showed up now and not on any of the latest PRs?

The black issue resulted from a change in one of its dependencies that was made March 28.  For the whole story see psf/black#2964.  It looks like many projects were affected!

> as long as the formatting behavior stays the same

Now that black is no longer beta, there is a stability policy that says the same output is produced for all releases in a calendar year.  If CQ does move to black 22.3.0 there will be some changes coming from 19.10b0.  I checked `cq.py` and 39 lines changed after formatting with black 22.3.0.

adam-urbanczyk, 2022-03-31:
I'm trying to pin click as suggested in the black issue

adam-urbanczyk, 2022-03-31:
#1018 works now

closed 2022-04-02 by lorenzncode