CadQuery/cadquery #1038: Introducing the VScode extension for CadQuery

roipoussiere, 2022-03-31:
I created a VSCode extension for CadQuery:

[image]

It's available on the [Open VSX Registry](https://open-vsx.org/extension/roipoussiere/cadquery) and on the [VisualStudio marketplace](https://marketplace.visualstudio.com/items?itemName=roipoussiere.cadquery).

Features:
- update preview on save;
- reasonably fast (ie. a fraction of a second for basic models);
- reasonably easy to install;

It's currently in alpha and I need beta testers. I wonder if I can create a PR on this repo to briefly introduce the project, for instance on the readme (below the *CadQuery editor* section) and the documentation (at the bottom of the *Installing CadQuery* page).

This will increase the project visibility, and therefore I will have more feedback on it.

jmwright, 2022-04-01:
@roipoussiere Is this based on the work mentioned [here](https://github.com/bernhard-42/jupyter-cadquery/issues/40)? Is it based on jupyter-cadquery, or was it built from scratch? If this is added to the readme, I'm thinking right below the jupyter-cadquery entry.

I tried to use it for a short time and have two notes.
- I had to update to the latest version of VSCode to get the extension to work in Windows. I'm not sure why.
- Is there a way to run this extension with an Anaconda environment, or does it always require Docker?

roipoussiere, 2022-04-01:
> Is this based on the work mentioned https://github.com/bernhard-42/jupyter-cadquery/issues/40?

Yep, it's this one!

> Is it based on jupyter-cadquery, or was it built from scratch?

Its based on the jupyter-cadquery's viewer (which now lives in [its own repo](https://github.com/bernhard-42/three-cad-viewer)).

> I had to update to the latest version of VSCode to get the extension to work in Windows. I'm not sure why.

Yep, it should be fixed in the next release. The minimal required VSCode was automatically set to my current VSCode version by Yarn.

> Is there a way to run this extension with an Anaconda environment, or does it always require Docker?

For now, the docker container is mandatory, but I plan to release a pip package that do the server part, this way one could run the server in its own environment.

Thank for your feedback!

Oh, and since then, I added type checking and code completion (this works even if you don't have cq installed locally):

[image]

roipoussiere, 2022-04-02:
I published a new version: 0.1.1

It is now be possible to avoid the use of Docker, see [the documentation](https://open-vsx.org/extension/roipoussiere/cadquery#install-the-cadquery-server).

jmwright, 2022-04-19:
I tried to install the cadquery2 package via pip per the instructions and got this error. This seems related to the new wheel on PyPi.
```
ERROR: Could not find a version that satisfies the requirement ocp-vtk<8.0.0,>=7.5.3 (from cadquery2) (from versions: none)
ERROR: No matching distribution found for ocp-vtk<8.0.0,>=7.5.3 (from cadquery2)
```

roipoussiere, 2022-04-19:
Well, that strange.

Did you use a virtual environment? Which system and python version are you using?

Can you at least do `pip install ocp-vtk`?

fpq473, 2022-04-19:
The pip version may be relevant.  I can reproduce your error on Linux if I use the version of pip (21.1.3) that shipped with Python 3.9.5.  The error goes away after I upgrade pip.

jmwright, 2022-04-19:
Updating to pip 22.0.4 worked. I get an error that the cadquery module doesn't exist when I try `import cadquery` in the Python repl though. I ran `pip install cadquery2` and didn't see any errors.

jmwright, 2022-04-20:
I have everything installed now, but had to use the git pip installation method for cadquery as discussed on Discord.

It might be worth noting in the documentation that the user has to use the jupyter-cadquery syntax of `show` instead of `show_object`. That had me stumped for a little bit.

This looks good. Thanks for all the work on it.

jmwright, 2022-04-20:
I just noticed this. Is there something extra that needs to be done to get code completion working? It's not working for me.

Mootch, 2022-04-29:
Hi. I manually add the stub path to Pylance and code completion seems to be working. However, code completion stops after any instance of finalize() in the code.

qkum, 2022-05-07:
> The pip version may be relevant. I can reproduce your error on Linux if I use the version of pip (21.1.3) that shipped with Python 3.9.5. The error goes away after I upgrade pip.

i just got the same error, and as you said, `conda install pip` (installing the latest) fixed the "pip cant find ocp-vtk" error. (i prefer conda virtual environments)

qkum, 2022-05-07:
> 

can you please elaborate on what ` use the git pip installation method for cadquery` is. 

i'm stuck at the same problem.

`(Cad_Query_Server) XXX@YYY:~/ZZZ/Cad_Query$ cq-server
Traceback (most recent call last):
  File "/home/XXX/anaconda3/envs/Cad_Query_Server/bin/cq-server", line 5, in <module>
    from src.server import main
  File "/home/XXX/anaconda3/envs/Cad_Query_Server/lib/python3.9/site-packages/src/server.py", line 3, in <module>
    import cadquery
ModuleNotFoundError: No module named 'cadquery'
`

qkum, 2022-05-07:
> > 
> 
> can you please elaborate on what ` use the git pip installation method for cadquery` is sir.
> 
> i'm stuck at the same problem.
> 
> `(Cad_Query_Server) XXX@YYY:~/ZZZ/Cad_Query$ cq-server Traceback (most recent call last): File "/home/XXX/anaconda3/envs/Cad_Query_Server/bin/cq-server", line 5, in <module> from src.server import main File "/home/XXX/anaconda3/envs/Cad_Query_Server/lib/python3.9/site-packages/src/server.py", line 3, in <module> import cadquery ModuleNotFoundError: No module named 'cadquery' `

ok, i found a solution that works if you use anaconda.
from the `https://cadquery.readthedocs.io/en/latest/installation.html` anaconda install guide.

**Install Latest Cad Query Version:**
`conda install -c cadquery -c conda-forge cadquery=master`

Cylon-Garage, 2022-07-05:
This extension is AMAZING! I am shocked at the low number of downloads of the extension from the VSCode marketplace. I'm doing research in a lab and will be showing all my colleges this tool!

roipoussiere, 2022-09-12:
Hello there!

The VSCode extension project is going to be replaced by cadquery-server: https://github.com/roipoussiere/cadquery-server
It works well with VSCode and has other useful features.

I created a similar issue to introduce it: https://github.com/CadQuery/cadquery/issues/1147

I'm now closing this, enjoy! :)

closed 2022-09-12 by roipoussiere