CadQuery/cadquery #1040: Tagged face doesn't follow object

mrtodd, 2022-04-01:
I was surprised by the behavior illustrated by the following code block:

``template = (cq.Workplane()
            .box(4,5,6)
            .faces(">X").tag("text_loc").end()
            .faces(">Y")
            .box(3,4,9)
            )

part1 = (template
         .faces(tag="text_loc")
         .workplane()
         .text("part1",1,-0.2)
         )

part2 = (template.translate([0,0,10])
         .faces(tag="text_loc")
         .workplane()
         .text("part2",1,-0.2)
         )

show_object(part1)
show_object(part2)
show_object(part2.faces(tag="text_loc"))
``
part2 doesn't get the text applied because the tagged face remains at the original location after the +10 Z translation of the template object. (You may want to comment out the first show_object call to see this.)

My expectation was that the tagged object would reflect any location/orientation changes made to the part that contains it.  I'm new.  Perhaps I don't understand something very basic about the tool.

Thanks in advance for any comments about this.

-- Todd

lorenzncode, 2022-04-08:
`translate` _Returns a copy of all of the items on the stack moved by the specified translation vector._

`translate` will move (a copy of) the item in blue here, a Compound in this case.  It does not move items on other Workplane objects.  There is more info in the docs on the stack, and Workplane chaining in the _CadQuery Concepts_ section. 

[image]

mrtodd, 2022-04-08:
So the question is why is the tagged object created isn't included by the compound object?  It was brought into existence within the context of the template object.

I could have done something like:
```
template = cq.Workplane().box(4,5,6)
text_loc = template.faces(">X")
template = template.faces(">Y").box(3,4,9)
```
to create a separate object.  This is not what I wanted -- I wanted a handle or "tag" to a specific geometry within the template object (in order to do something with it later).  In the code I am trying to write I have multiple tagged geometries in a "template"-like object which gets placed multiple times with differing locations/orientations in a larger part.  It would be nice not to have to move/reorient each of these individual tagged items for every instance in which the "template" object is used.

I originally was using something like 'faces("<Z[1]")' to access a desired geometry, but the template is parameterized and the selector didn't always return the geometry I wanted because with some parameter values other faces could be created above the one I was seeking.  I do always know which face I want to access later when the template is created so it seemed applying a tag at the time the geometry is created was the perfect solution.

Is there a way to get this functionality?

mrtodd, 2022-04-08:
I tried the following:
```
t2 = (cq.Workplane() #origin=cq.Vector(10,0,0))
      .box(4,5,6).faces(">X").tag("text_loc").end()
      .faces(">Y")
      .box(3,4,9)
      )
show_object(t2,options={"color":"purple"})

tf = t2.faces(tag="text_loc")
t2 = t2.add(tf)
t2 = t2.translate([0,0,10])

# show_object(t2)
show_object(t2.faces(tag="text_loc"),options={"color":"orange"})
show_object(t2.objects[0],options={"color":"green"})
show_object(t2.objects[1],options={"color":"pink"})
```
The tagged face can be included in the template, but the tag still points to the original location.  How can one make the face move and maintain a tag that points to it
[image]
?

jmwright, 2022-04-11:
@mrtodd The tag isn't modified when the copy happens while doing the translate. As you have discovered, it still points to the original object. This probably isn't what you want, but translating afterward will put the text in the correct place using the tagged face.
```python
import cadquery as cq

template = (cq.Workplane()
.box(4,5,6)
.faces(">X").tag("text_loc").end()
.faces(">Y")
.box(3,4,9)
)

part1 = (template
.faces(tag="text_loc")
.workplane()
.text("part1",1,-0.2)
)

part2 = (template
.faces(tag="text_loc")
.workplane()
.text("part2",1,-0.2)
.translate([0,0,10])
)

show_object(part1)
show_object(part2)
```
[image]

mrtodd, 2022-04-11:
@jmwright   Thanks for your response.  I was already aware of this solution, but it really isn't a good fit with what I am trying to do.  I think I might (try to) write some code to override the definitions for 'translate', 'rotate' and any other methods for cq.Workplane() that deal with changing location or orientation of an object so they operate on everything tagged in the target object as well as the object itself.

mrtodd, 2022-04-12:
Adding the following seems to fix my problem for rotate and translate:

```
def tmk_rotate(
    self: T,
    axisStartPoint: VectorLike,
    axisEndPoint: VectorLike,
    angleDegrees: float,
    ) -> T:
    """
    Returns a copy of all of the items on the stack rotated through and angle around the axis
    of rotation.

    :param axisStartPoint: The first point of the axis of rotation
    :type axisStartPoint: a 3-tuple of floats
    :param axisEndPoint: The second point of the axis of rotation
    :type axisEndPoint: a 3-tuple of floats
    :param angleDegrees: the rotation angle, in degrees
    :type angleDegrees: float
    :returns: a CQ object
    """
    t_list = self.ctx.tags.keys()
    for t in t_list:
        t_obj = self.ctx.tags[t]
        self.ctx.tags[t] = t_obj.newObject(
            [
                o.rotate(Vector(axisStartPoint),
                         Vector(axisEndPoint),
                         angleDegrees
                         )
                if isinstance(o, Shape)
                else o
                for o in t_obj.objects
            ]
        )
    return self.newObject(
        [
            o.rotate(Vector(axisStartPoint), Vector(axisEndPoint), angleDegrees)
            if isinstance(o, Shape)
            else o
            for o in self.objects
        ]
    )
cq.Workplane.rotate = tmk_rotate

def tmk_translate(self: T, vec: VectorLike) -> T:
    """
    Returns a copy of all of the items on the stack moved by the specified translation vector.

    :param tupleDistance: distance to move, in global coordinates
    :type  tupleDistance: a 3-tuple of float
    :returns: a CQ object
    """
    t_list = self.ctx.tags.keys()
    for t in t_list:
        t_obj = self.ctx.tags[t]
        self.ctx.tags[t] = t_obj.newObject(
            [
                o.translate(Vector(vec)) 
                if isinstance(o,Shape) else o
                for o in t_obj.objects
            ]
        )
    return self.newObject(
        [
            o.translate(Vector(vec)) if isinstance(o, Shape) else o
            for o in self.objects
        ]
    )
cq.Workplane.translate = tmk_translate

```
I haven't yet looked into which other methods would need changes made (mirror is the next one in cq.py and I assume it would require similar changes).

Is this something that makes sense to include in the tool or am I the only one who sees this as useful?  Can anyone think of problems this might cause which I have not yet uncovered?

jmwright, 2022-04-12:
> Is this something that makes sense to include in the tool or am I the only one who sees this as useful? Can anyone think of problems this might cause which I have not yet uncovered?

It seems like it would be confusing for the majority of users to have their tagged object moving around when certain operations are performed. At the very least, it would need to be an opt-in parameter for any functions that transformed the tag to avoid confusion and retain backwards compatibility.

mrtodd, 2022-04-12:
I'm guessing that if users found the tag mechanism useful before, it is because they always made reference to tagged objects before any change in location or orientation to the object containing the tag was made.  Therefore, they wouldn't even be aware that the tagged object had been altered.

adam-urbanczyk, 2022-04-14:
You are proposing a fundamental and breaking change, so thanks but no thanks.

closed 2022-04-14 by adam-urbanczyk

mrtodd, 2022-04-14:
That's fine, but the tag mechanism "as is" is completely useless. It does nothing but allow one to just continue on with the ugly run-on sentences.  Easily replaced by doing something like this example I showed before:
```
template = cq.Workplane().box(4,5,6)
text_loc = template.faces(">X")
template = template.faces(">Y").box(3,4,9)
```
What I proposed is actually more in line with what most cad systems do.  For example, look at integrated circuit layout where one tags a geometry in a cell template to tell the router where to make a connection.  That template gets instantiated many times and the router sees the connection geometries resolved to the location and orientation of each of the cell instantiations.

0xorial, 2024-08-24:
I have a similar problem. I am creating multiple instances of an somewhat complex object. at the creation time it is easy to locate specific face, which I want to use (in this case loft) later. However, after instantiations and transformation, there is no simple selector to locate this face. I realize that just mutating state of previous steps is not inline with fluent chained calls approach, but perhaps there is a less disruptive approach to achieve the same?..

0xorial, 2024-08-24:
Actually, thinking about it a more, wouldn't it make sense to clone the context on translate/rotate? this way tagged objects will behave more intuitively and there will be no impact on previously acquired references.