CadQuery/cadquery #1042: Importing STEP models on top of transformed workplanes

r4dr3fr4d, 2022-04-04:
I'd like to import a STEP onto a transformed/rotated workplane, instead of importing it and redoing previous transformations.  Is this possible?

gumyr, 2022-04-07:
The `importStep()` method returns the object as:
```python
    return cq.Workplane("XY").newObject(solids)
```
so I believe you'll need to translate/rotate the Shape afterwards.  

If it's useful to you cq_warehouse contains a `Shape.transformed()` method that enables translation and rotation in one operation just as the `Plane.transformed()` method (see https://github.com/gumyr/cq_warehouse/blob/main/src/cq_warehouse/extensions.py#L2049).

r4dr3fr4d, 2022-04-07:
I'm aware that you can transform it after importing. I was wondering if there existed a way to import it onto the current workplane of an object.

jmwright, 2022-04-11:
@r4dr3fr4d There is no way to currently do this. The code assumes a standard `XY` workplane when importing. [Reference](https://github.com/CadQuery/cadquery/blob/3f6da6d8b528d1556382e691b92b70a8e6c5d1b6/cadquery/occ_impl/importers/__init__.py#L67)