CadQuery/cadquery #1046: Error when importing cadquery: nptyping.error.InvalidArgumentsError

roipoussiere, 2022-04-08:
When trying to import cadquery I get the following error:

```
>>> import cadquery
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/seth/miniforge/envs/osr/lib/python3.9/site-packages/cadquery/__init__.py", line 14, in <module>
    from .occ_impl import exporters
  File "/home/seth/miniforge/envs/osr/lib/python3.9/site-packages/cadquery/occ_impl/exporters/__init__.py", line 10, in <module>
    from ...cq import Workplane
  File "/home/seth/miniforge/envs/osr/lib/python3.9/site-packages/cadquery/cq.py", line 61, in <module>
    from .sketch import Sketch
  File "/home/seth/miniforge/envs/osr/lib/python3.9/site-packages/cadquery/sketch.py", line 28, in <module>
    from .occ_impl.sketch_solver import (
  File "/home/seth/miniforge/envs/osr/lib/python3.9/site-packages/cadquery/occ_impl/sketch_solver.py", line 236, in <module>
    class SketchConstraintSolver(object):
  File "/home/seth/miniforge/envs/osr/lib/python3.9/site-packages/cadquery/occ_impl/sketch_solver.py", line 265, in SketchConstraintSolver
    self, x0: Array[(Any,), float]
  File "/home/seth/miniforge/envs/osr/lib/python3.9/site-packages/nptyping/ndarray.py", line 73, in __getitem__
    shape_expression, dtype = _get_from_tuple(item)
  File "/home/seth/miniforge/envs/osr/lib/python3.9/site-packages/nptyping/ndarray.py", line 128, in _get_from_tuple
    raise InvalidArgumentsError(
nptyping.error.InvalidArgumentsError: Unexpected argument '(typing.Any,)', expecting Shape[<ShapeExpression>] or Literal[<ShapeExpression>]
```

This error is really recent, because there was no problem to install and import cadquery yesterday.

I tried on old commits (for instance 3efb080 published 3 weeks ago) and I get the same error.
The suspicious line in the stacktrace is [here](https://github.com/CadQuery/cadquery/blob/master/cadquery/occ_impl/sketch_solver.py#L265).

I don't use Conda but pip, using the [ocpvtk wheel](https://gist.github.com/fpq473/9392185d722c8be722f61e4d745723ee) I built yesterday, but someone else got the same error using Conda.

If I use the python cadquery venv I created yesterday I can import cadquery,
But if I create a new venv (using the same ocpvtk wheel), I get `ModuleNotFoundError: No module named 'typish'` when trying to import cadquery.
Then if I install typish module and try again, I get the error described above. 
It's strange, because according to a [git blame on the line that imports typish](https://github.com/CadQuery/cadquery/blame/d4b5eec21b4bbec47712084b9f88d43fc8f6c165/cadquery/sketch.py#L19), it was added 5 months ago.

I compared the output of `pip freeze` of the cq venv I created yesterday and the venv I created today (just before adding typish): each dependency version are identical (as well as Python and pip version).

roipoussiere, 2022-04-08:
Ok, the issue appears because of the new release of np-typing (2.0) [published yesterday](https://github.com/ramonhagenaars/nptyping/releases/tag/v2.0.0).

I can fix the error by specifying `nptyping==1.4.4`.

sethfischer, 2022-04-08:
A work-around until this is fixed in master. When following [Command Line Installation](https://cadquery.readthedocs.io/en/latest/installation.html#command-line-installation), specify nptyping 1.4.4.

```
conda install -c cadquery -c conda-forge cadquery=master nptyping=1.4.4
```

lorenzncode, 2022-04-09:
Is it preferred to make changes to support nptyping v2.0.0?

jmwright, 2022-04-10:
@lorenzncode Supporting nptyping 2.x would probably be ideal, but we have to make sure that it doesn't cause problems anywhere else in CQ's dependency tree as well. If it does cause other problems, then we'll have to pin the version to 1.4.4.

closed 2022-04-11 by adam-urbanczyk