CadQuery/cadquery #1053: Create a discord server dedicated to CadQuery?

roipoussiere, 2022-04-14:
There is already a CadQuery channel on the Elmer discord, and I already like this place*, but I feel a bit cramped and as the CadQuery community is growing, I think that it deserves a better place.

Creating a discord server dedicated to CadQuery will allow the possibility to split discussions about cadquery in several channels, which is interesting compared to having a unique discussion flow. Some benefits are:
- more organized discussions;
- avoid drowning out current discussions;
- possibility to get notified to specific topic;
- easier to find old messages;
- etc.

Some channels suggestion:

- announcements about Cadquery;
- cadquery modules showcase;
- cadquery models showcase;
- requests for help on cadquery code;
- cadquery editor;
- cadquery vscode extension;
- issues;
- cadquery development;
- ...

I'd like to hear your opinion on this. :slight_smile:

_*_ putting aside the fact that I don't really understand why this channel belongs to a [multiphysical simulation software](http://www.elmerfem.org/) :thinking:

jmwright, 2022-04-15:
The Elmer mods have already hinted that they see themselves as more of an incubator for Discord communities than a permanent home.

I'm willing to set up a Discord server for CQ, but I have a hard time keeping up in the one channel we have now. We'd need some volunteer mods. There are already some likely candidates that are active on the Discord channel now and are active in the broader community. I'd probably DM some of them to see if they'd be willing to be mods.

We would also need to set the Matrix bridge back up as discussed [here](https://groups.google.com/g/cadquery/c/z3GmFj5pSzQ/m/i2yXAiJdAwAJ). We'd need to ask for help from the Matrix side again.

I'm not sure how to cleanly migrate users to a new Discord server. I think that all we can do is post the invite link, encourage people to move over, and then ask the Elmer admins archive the CadQuery channel on their server after a reasonable amount of time.

Jojain, 2022-04-15:
I second this, I think it would be easier to manage than the one discussion feed we have right now. I don't know if I was on the list of potential mods you were thinking about but I would be up to be a mod of the new channel !

jmwright, 2022-04-15:
I have set up a server and will start inviting the core devs and a few mods before we open it up all the way. I left development activities out from the suggested channel list above because we don't want to fragment those discussions between Discord and GitHub. I also created a category called "other-guis" as a catch-all because we have multiple GUIs that have been developed in addition to CQ-editor. My thought is that if any of the other environments (jupyter-cadquery, vscode-extension, Nales) start overloading the channel, we can always spin them off into their own channels later.

jmwright, 2022-04-19:
We will be on both the new and old servers for awhile, but a dedicated CadQuery server is now available. The Matrix bridge is not in place, but I am not able to get in contact with anyone on the Matrix side of the bridge. We can always add it back later. https://discord.gg/Bj9AQPsCfx

closed 2022-04-19 by jmwright