CadQuery/cadquery #1055: Allow STL export in binary format in addition to ASCII

shimwell, 2022-04-19:
Currently the CadQuery export STL feature makes use of OCP.StlAPI_Writer

This writes an ASCII STL file but it would be helpful (to preserve high precision) to be able to write a binary STL file.

Perhaps an option to the CadQuery export could be added that allows STL files to be saved in binary format, it looks like OCCT supports binary STL files

Links to OCCT documentation:
Convert and write shape to STL format. file is written in binary if aAsciiMode is False otherwise it is written in Ascii (by default) 
https://dev.opencascade.org/doc/occt-7.0.0/refman/html/class_stl_a_p_i.html

lorenzncode, 2022-04-23:
I tested exporting binary STL.

`exporters.export`  - added "ascii" to the opt dict.

https://github.com/lorenzncode/cadquery/blob/e53bdc0951e67da23cccf31f5b334022b23af654/cadquery/occ_impl/exporters/__init__.py#L105-L109

example usage:
`exporters.export(result, "test_binary.stl", "STL",0.05, 0.1, opt={"ascii": False})`

The change to `Shape.exportStl` signature was to add a bool parameter to control ASCII or binary output.
```
    def exportStl(
        self,
        fileName: str,
        tolerance: float = 1e-3,
        angularTolerance: float = 0.1,
        ascii: bool = True,
    ) -> bool:
```

example usage:
`result.val().exportStl("test_binary2.stl", 0.05, 0.1, False)`

Any feedback for/against adding this feature?

jmwright, 2022-04-28:
@lorenzncode I would be for adding it.

closed 2022-08-16 by jmwright