CadQuery/cadquery #1057: Strange offset2D result

mrtodd, 2022-04-20:
Any idea why offset2D gives different results for these two test blocks?

```
import cadquery as cq

test1 = (cq.Workplane()
        .box(25,25,1)
        .faces(">Z").workplane()
        .moveTo(7.825,-3.155)
        .circle(3.15/2).cutThruAll()
        .moveTo(-7.825,-3.155)
        .circle(3.15/2).cutThruAll()
        .moveTo(-7.825,3.155)
        .circle(3.15/2).cutThruAll()
        .moveTo(7.825,3.155)
        .circle(3.15/2).cutThruAll()
        )

test2 = (cq.Workplane()
         .box(25,25,1)
         .faces(">Z").workplane()
         .rect(2*7.825,2*3.155,forConstruction=True)
         .vertices().circle(3.15/2).cutThruAll()
         )

circles1 = test1.faces(">Z").edges("%CIRCLE")
small_circles1 = circles1.toPending().offset2D(-0.5)
# same as test1, but shifted up by 2 for viewing
circles2 = test2.translate([0,0,2]).faces(">Z").edges("%CIRCLE")
small_circles2 = circles2.toPending().offset2D(-0.5)

show_object(circles1,options={"color":"blue"})
show_object(small_circles1,options={"color":"red"})

show_object(circles2,options={"color":"orange"})
show_object(small_circles2,options={"color":"yellow"})

```
[image]

For test1 (red/blue) the offset is correct for the circle at the last 'moveTo' position (shuffle the order of the moveTo lines and the last one will have the proper offset)..

I also saw a case where the circles were sized properly, but their locations were shifted in X, Y and Z from where they were expected.  There are two other issues ( [#896](https://github.com/CadQuery/cadquery/issues/896) and [#753](https://github.com/CadQuery/cadquery/issues/753)) already open that sounded like they are related to that issue.

lorenzncode, 2022-04-21:
ShapeUpgrade_UnifySameDomain?  I find the red wire is inside with `cutThruAll(clean=False)`.

mrtodd, 2022-04-21:
Yes, that does fix it.
I also made the following change to the code where I encountered the shift in position in all three axis:
```
            # .rarray(ps,h_in_sp+d,1,2).hole(d,clean=False)
            .rarray(ps,h_in_sp+d,1,2).circle(d/2).cutThruAll(clean=False)

```
And, this resolved that problem as well.  I first added "clean=False" to the hole command and still got the incorrect result, then I commented out that line and replaced it with the line shown above.

mrtodd, 2022-04-21:
This might be helpful to someone trying to understand what is happening with the default clean=True for cutThruAll().

I'm just showing some snippets from the code which had the all-axis shifting problem (nobody would want to see the whole thing), but I hope to provide enough of a view to make it possible to understand.

```
 part = (cq.Workplane()
            .box(w,l,h)
            .faces(">Z").workplane(origin=cq.Vector(w/2-(h_ed_sp+d/2),0,0))
            # .rarray(ps,h_in_sp+d,1,2).hole(d,clean=False)
            .rarray(ps,h_in_sp+d,1,2).circle(d/2).cutThruAll(clean=False)
            .faces(">Z").edges("%CIRCLE").tag("holes").end()
            .faces(">Z").workplane(origin=cq.Vector((ps-w)/2,0,0))
            .rarray(ps,ps,1,8).hole(pd,clean=False)
            )
```
Then a few hundred lines later I access the tagged edges using this code:

```
holes = mpu.edges(tag="holes")  #.toPending().offset2D(0)
holes_small = holes.toPending().offset2D(-0.5)
```
The length of holes.objects is 6 for the two circles.  When run with the ".toPending().offset2D(0)" not commented out the list is length 2.  It appears that running offset2D with an offset of zero does what clean failed to do correctly inside cutThruAll().