CadQuery/cadquery #1065: Please add more examples

nesdnuma, 2022-04-27:
Hello,

Please add more examples, especially examples dealing with construction of multiple objects through a for loop.

Thank you.

tommy3001, 2022-04-28:
What do you mean with multiple objects? 

In case of assemblies, you can do it in the following way: 
```
import cadquery as cq

beam = [120,80,3000];

beams= cq.Assembly()

def f_beams(b_offset):
   
    beams.add(cq.Workplane("XY").box(beam[0],beam[1],beam[2]),loc=cq.Location(cq.Vector(0,b_offset*500,0)))

for a in range(0,9): 
    f_beams(a)
  
show_object(beams)

```

Additionally, you should read the cadquery concepts

https://cadquery.readthedocs.io/en/latest/primer.html#

dcowden, 2022-04-28:
@nesdnuma could you provide a list of a few modules/situations you would have found useful ?  I'm afraid a lot of us who know it well are 'nose-blind' to what things are obvious and which ones are not at this point.  It would go a long way if you could list the concepts/functions that were confusing as a beginner.

nesdnuma, 2022-04-28:
Actually it's hard to tell, I am too much of a beginner, also with Python. I am bumping on new issues all the time. The ideal situation would be a code example for each module/function described in the API but I understand this would be a lot of work. I'll try to give some examples myself as soon as I have something of interest.

nesdnuma, 2022-05-03:
An example including sweep() would be nice.

nesdnuma, 2022-05-13:
Also, examples of the various arc methods:
Workplane.threePointArc
Workplane.sagittaArc
Workplane.radiusArc
Workplane.tangentArcPoint

nesdnuma, 2022-05-17:
Examples with Workplane.wire(), Workplane.wires(), Workplane.parametricCurve()...

adam-urbanczyk, 2022-05-17:
In the meantime you can check the tests.

nesdnuma, 2022-05-19:
> In the meantime you can check the tests.

Which tests???

jmwright, 2022-05-19:
@nesdnuma [Here](https://github.com/CadQuery/cadquery/tree/master/tests).

Probably the best way to go about it is to search CadQuery's GitHub repo for the topic and then click on any test entries that come up. For instance, if you search `revolve` there is an entry under `test_cadquery.py` in the results that says `testRevolveCylinder`. You can click to go to that line of code and look at how `revolve` was used.

nesdnuma, 2022-05-19:
Interesting! Thank you @jmwright !