CadQuery/cadquery #1068: Import CadQuery into Blender with pip

lawrence-dubowitz, 2022-04-27:
How would I go about installing CadQuery into Blenders preinstalled python distribution (3.10.2) with pip?

I want to integrate CadQuery into a Blender plugin for parametric modelling with geometry nodes similar to Rhino's Grasshopper.

Thanks in advance!

jmwright, 2022-04-27:
I'm not sure how it works with Blender, but in the Godot-Python plugin you use the pip binary that's bundled with the plugin and the package installs into the directory structure of the plugin. So something like `/path/to/plugin/pip install cadquery` Are there instructions on how to install other PyPi packages in Blender that you can link to?

This is a temporary package while we work on getting the Python wheels onto our main PyPi account, but you can try installing CadQuery with `pip install cadquery2` for now.

I'm tagging @roipoussiere on this as well.

lawrence-dubowitz, 2022-04-27:
thanks for the quick response!

what other dependencies do I need to install manually? Its complaining that it cant find the module OCP...

jmwright, 2022-04-27:
Doing `pip install cadquery2` should have installed the ocp-vtk wheel automatically, but that install process is fairly new, so there could be a bug in this use case. You could try `pip install ocp-vtk` separately to see what happens.

lawrence-dubowitz, 2022-04-28:
Unfortunately no luck installing ocp-vtk. I'll continue trying to get it to work and I'll post the solution if I manage to get it working.

jmwright, 2022-04-28:
@lawrence-dubowitz It seems like there might be an issue with the ocp-vtk wheel in Windows 10, so it will take time to fix that. You could try creating an Anaconda environment and then launching Blender from the command line with that environment activated. I'm not sure that will work for Blender, but I used to run Godot that way in Linux.

lawrence-dubowitz, 2022-04-28:
Thanks, I'll give it a try and let you know how it goes...

jmwright, 2022-05-05:
@lawrence-dubowitz The following should work as a way to test (it worked for me on Windows 11 and Python 3.8), and the community is working on a permanent solution that works seamlessly with PyPi.

If Blender is running Python 3.8:
```
pip install https://github.com/jmwright/ocp-build-system/releases/download/7.5.3/ocp_vtk-7.5.3.0-cp38-cp38-win_amd64.whl 'git+https://github.com/CadQuery/cadquery' ezdxf multimethod nlopt nptyping typish
```

If Blender is running Python 3.9:
```
pip install https://github.com/jmwright/ocp-build-system/releases/download/7.5.3/ocp_vtk-7.5.3.0-cp39-cp39-win_amd64.whl 'git+https://github.com/CadQuery/cadquery' ezdxf multimethod nlopt nptyping typish
```

If Blender is running Python 3.10:
```
pip install https://github.com/jmwright/ocp-build-system/releases/download/7.5.3/ocp_vtk-7.5.3.0-cp310-cp310-win_amd64.whl 'git+https://github.com/CadQuery/cadquery' ezdxf multimethod nlopt nptyping typish
```

lawrence-dubowitz, 2022-05-05:
Wonderful, thank you very much!

I will give it a try as soon as I get a chance...

jmwright, 2022-05-23:
@lawrence-dubowitz We'll have a release on PyPi sometime in the near future, but you can now install the latest CadQuery commit via pip from the git repo. You'll need a fairly recent version of pip first.
```
python3 -m pip install --upgrade pip
```
And then you can run this line.
```
pip install git+https://github.com/CadQuery/cadquery.git
```
Let us know if this works in Blender.

lawrence-dubowitz, 2022-05-28:
Unfortunately it failed with the following output when installing and
trying to use in blender:

*C:\Program Files\Blender Foundation\Blender 3.1\3.1\python\bin>python.exe
>> -m pip install git+https://github.com/CadQuery/cadquery.git
>> <https://github.com/CadQuery/cadquery.git>*
>
> *WARNING: Ignoring invalid distribution -umpy (c:\program files\blender
>> foundation\blender 3.1\3.1\python\lib\site-packages)*
>
> *WARNING: Ignoring invalid distribution -umpy (c:\program files\blender
>> foundation\blender 3.1\3.1\python\lib\site-packages)*
>
> *Collecting git+https://github.com/CadQuery/cadquery.git
>> <https://github.com/CadQuery/cadquery.git>*
>
> *  Cloning https://github.com/CadQuery/cadquery.git
>> <https://github.com/CadQuery/cadquery.git> to
>> c:\users\lawre\appdata\local\temp\pip-req-build-1npwi9hy*
>
> *  Running command git clone -q https://github.com/CadQuery/cadquery.git
>> <https://github.com/CadQuery/cadquery.git>
>> 'C:\Users\lawre\AppData\Local\Temp\pip-req-build-1npwi9hy'*
>
> *  Resolved https://github.com/CadQuery/cadquery.git
>> <https://github.com/CadQuery/cadquery.git> to commit
>> 3032e0f8fe4856b037313def4b0bf431cae87708*
>
> *Collecting cadquery-ocp*
>
> *  WARNING: Retrying (Retry(total=4, connect=None, read=None,
>> redirect=None, status=None)) after connection broken by
>> 'ReadTimeoutError("HTTPSConnectionPool(host='files.pythonhosted.org
>> <http://files.pythonhosted.org>', port=443): Read timed out. (read
>> timeout=15)")':
>> /packages/e6/85/e4231752aacc17cb8f6233f4859b549cac5964c21ee8836c92c883e40504/cadquery_ocp-7.5.3-cp310-cp310-win_amd64.whl*
>
> *  Downloading cadquery_ocp-7.5.3-cp310-cp310-win_amd64.whl (85.3 MB)*
>
> *     |████████████████████████████████| 85.3 MB 595 kB/s*
>
> *Requirement already satisfied: ezdxf in c:\program files\blender
>> foundation\blender 3.1\3.1\python\lib\site-packages (from
>> cadquery==2.2.dev454+g3032e0f) (0.17.2)*
>
> *Requirement already satisfied: multimethod in c:\program files\blender
>> foundation\blender 3.1\3.1\python\lib\site-packages (from
>> cadquery==2.2.dev454+g3032e0f) (1.8)*
>
> *Requirement already satisfied: nlopt in c:\program files\blender
>> foundation\blender 3.1\3.1\python\lib\site-packages (from
>> cadquery==2.2.dev454+g3032e0f) (2.7.1)*
>
> *Requirement already satisfied: nptyping>=2 in c:\program files\blender
>> foundation\blender 3.1\3.1\python\lib\site-packages (from
>> cadquery==2.2.dev454+g3032e0f) (2.0.0)*
>
> *Collecting typish*
>
> *  Using cached typish-1.9.3-py3-none-any.whl (45 kB)*
>
> *Collecting casadi*
>
> *  WARNING: Retrying (Retry(total=4, connect=None, read=None,
>> redirect=None, status=None)) after connection broken by
>> 'ReadTimeoutError("HTTPSConnectionPool(host='files.pythonhosted.org
>> <http://files.pythonhosted.org>', port=443): Read timed out. (read
>> timeout=15)")':
>> /packages/16/07/867767f011cd92b79da421a774fd8305fb3e69fd342b79fea990a95a9b8a/casadi-3.5.6.pre2-cp310-none-win_amd64.whl*
>
[... 90 lines cut ...]
  File "<blender_console>", line 1, in <module>
  File "C:\Program Files\Blender Foundation\Blender
3.1\3.1\python\lib\site-packages\cadquery\__init__.py", line 22, in <module>
    from .occ_impl import exporters
  File "C:\Program Files\Blender Foundation\Blender
3.1\3.1\python\lib\site-packages\cadquery\occ_impl\exporters\__init__.py",
line 10, in <module>
    from ...cq import Workplane
  File "C:\Program Files\Blender Foundation\Blender
3.1\3.1\python\lib\site-packages\cadquery\cq.py", line 61, in <module>
    from .sketch import Sketch
  File "C:\Program Files\Blender Foundation\Blender
3.1\3.1\python\lib\site-packages\cadquery\sketch.py", line 28, in <module>
    from .occ_impl.sketch_solver import (
  File "C:\Program Files\Blender Foundation\Blender
3.1\3.1\python\lib\site-packages\cadquery\occ_impl\sketch_solver.py", line
10, in <module>
    import nlopt
  File "C:\Program Files\Blender Foundation\Blender
3.1\3.1\python\lib\site-packages\nlopt\__init__.py", line 1, in <module>
    from .nlopt import *
  File "C:\Program Files\Blender Foundation\Blender
3.1\3.1\python\lib\site-packages\nlopt\nlopt.py", line 20, in <module>
    from . import _nlopt
ImportError: numpy.core.multiarray failed to import

jmwright, 2022-05-28:
Looks like Blender may be bundling an older version of Python than we can support with the wheels at this time. What does `C:\Program Files\Blender Foundation\Blender 3.1\3.1\python\bin>python.exe` show?

lawrence-dubowitz, 2022-05-29:
*C:\Program Files\Blender Foundation\Blender 3.1\3.1\python\bin>python.exe*
*Python 3.10.2 (main, Jan 27 2022, 08:34:43) [MSC v.1928 64 bit (AMD64)] on
win32*
*Type "help", "copyright", "credits" or "license" for more information.*
*>>>*

jmwright, 2022-05-30:
The installation works for me if I force pip to run with the bundled version of Python.
```
"C:\Program Files\Blender Foundation\Blender 3.1\3.1\python\bin\python.exe" -m pip install git+https://github.com/CadQuery/cadquery.git
```
I didn't even upgrade pip since it was already at version 21, which is good enough to install our packages.
```
"C:\Program Files\Blender Foundation\Blender 3.1\3.1\python\bin\python.exe" -m pip --version
```
 There were a couple of warnings during the install, but they didn't keep CadQueryfrom working. I was able to do `import cadquery` after running python.exe to get the REPL.

jmwright, 2022-05-30:
@lawrence-dubowitz Following up on the last comment. Are you running the command from cmd or PowerShell? I ran mine from cmd, and although I didn't run it in administrator mode, that might be worth a try.

I show the same Python version and platform information on my Windows system that you have, so I expect there's something minor that is still keeping this from working.

lawrence-dubowitz, 2022-05-30:
Thanks, I'm still struggling to get it working with cmd (also with admin
privileges), but at least I know that it's working on your side, so it must
be something on my side. I will persist and update the thread once I find
out what's going on.

Everything seems to be installing correctly, but I get the following error
when running 'import cadquery' inside the Blender REPL:

*>>> import cadquery RuntimeError: module compiled against API version 0xf
but this version of numpy is 0xe Traceback (most recent call last): File
"<blender_console>", line 1, in <module> File "C:\Program Files\Blender
Foundation\Blender 3.1\3.1\python\lib\site-packages\cadquery\__init__.py",
line 22, in <module> from .occ_impl import exporters File "C:\Program
Files\Blender Foundation\Blender
3.1\3.1\python\lib\site-packages\cadquery\occ_impl\exporters\__init__.py",
line 10, in <module> from ...cq import Workplane File "C:\Program
Files\Blender Foundation\Blender
3.1\3.1\python\lib\site-packages\cadquery\cq.py", line 61, in <module> from
.sketch import Sketch File "C:\Program Files\Blender Foundation\Blender
3.1\3.1\python\lib\site-packages\cadquery\sketch.py", line 28, in <module>
from .occ_impl.sketch_solver import ( File "C:\Program Files\Blender
Foundation\Blender
3.1\3.1\python\lib\site-packages\cadquery\occ_impl\sketch_solver.py", line
10, in <module> import nlopt File "C:\Program Files\Blender
Foundation\Blender 3.1\3.1\python\lib\site-packages\nlopt\__init__.py",
line 1, in <module> from .nlopt import * File "C:\Program Files\Blender
Foundation\Blender 3.1\3.1\python\lib\site-packages\nlopt\nlopt.py", line
20, in <module> from . import _nlopt ImportError: numpy.core.multiarray
failed to import*

jmwright, 2022-05-30:
Is there anything special about your Blender installation? What's your full Blender version number? I installed it this morning on a Windows system that's never had Blender installed before. I downloaded it from [here](https://www.blender.org/download/) (version 3.1.2). You could start the Python REPL and try the following just to see if at least OCP is working: `from OCP import *`. If OCP is not installed, you can do `pip.exe install cadquery-ocp`.

You could also try `pip.exe install numpy --upgrade` but I don't know if that will break other things in your Blender install. Try at your own risk.