CadQuery/cadquery #1069: Assemble VRML export incorrect rotation

lorenzncode, 2022-04-28:
From the docs cone example:
```
import cadquery as cq

cone = cq.Solid.makeCone(1, 0, 2)

assy = cq.Assembly()
assy.add(cone, name="cone0", color=cq.Color("blue"))
assy.add(
    cone,
    loc=cq.Location(cq.Vector(0, 0, 0), cq.Vector(1, 0, 0), 180),
    name="cone1",
    color=cq.Color("green"),
)

show_object(assy)

assy.save("out_assy.wrl", "VRML")
cq.exporters.assembly.exportVRML(assy, "out_assy2.wrl")
cq.exporters.export(assy.toCompound(), "out_compound.wrl", "VRML")
```

CQ-Editor view is on the right, with VRML viewer overlay on left:
[image]

I find that this issue is due to radians/degrees mismatch in `toVTK`.

jmwright, 2022-04-28:
@lorenzncode So does the conversion need to be one on `rot` [here](https://github.com/CadQuery/cadquery/blob/3581904614b1aa31d0deea4e8cfcf4771e5e1943/cadquery/occ_impl/assembly.py#L185)?

lorenzncode, 2022-04-28:
Yes, the VTK function expects degrees - confirmed with VTK docs and testing change of `toVTK`.

https://vtk.org/doc/nightly/html/classvtkProp3D.html

PR #1078 (2022-05-04): Fix VTK based assembly export rotation issue
To resolve #1069.  As a test, I'm checking a line of the VRML file itself.  The file is less than 50 lines.
```
#VRML V2.0 utf8
# VRML file written by the visualization toolkit

    Background {
    skyColor [1.000000 1.000000 1.000000, ]
    }
     Viewpoint
      {
      fieldOfView 0.523599
      position -1.000000 0.000000 1.931852
      description "Default View"
      orientation 0 0 1 0
      }
    NavigationInfo {
      type ["EXAMINE","FLY"]
      speed 4.000000
      headlight TRUE}

    DirectionalLight { ambientIntensity 1 intensity 0 #  [... cut]
closed 2022-05-04 by adam-urbanczyk