CadQuery/cadquery #1080: extrude(until="next") shuld respect non filled faces

qwelyt, 2022-05-05:
```python
box = (cq.Workplane("XY")
       .box(10,10,10)
       .faces(">Z")
       .shell(1)
       )
box_with_sketch = (box.faces(">Z")
                   .sketch()
                   .circle(5)
                   .rect(1,4, mode="s")
                   .rect(4,1, mode="s")
                   .finalize()
                   )
extrude_value = box_with_sketch.extrude(-10)
extrude_next = box_with_sketch.extrude(until="next")
show_object(extrude_value)
show_object(extrude_next.translate((13,0,0)))
```

I would expect both these boxes to look the same. But the one using `until="next"` ignores the cross and extrudes a fully filled circle.

[image]

Please make `until="next"` respect the form that is being extruded. (It seems that occt does not really say if this is supported or not https://dev.opencascade.org/doc/refman/html/class_b_rep_feat___make_d_prism.html but could perhaps be solved by CQ?)

adam-urbanczyk, 2022-05-06:
Hm, looks like a limitation of `dprism`

Jojain, 2022-05-06:
One thing that can be done is to add logic in the extrude method to make two dprism and cut the inner dprism from the outer one