CadQuery/cadquery #1081: Hello! Question about filleting specific edges (selecting edges by index)

TristanAntonsen, 2022-05-06:
I come from the engineering CAD world and I'm creating some designs in CadQuery. What I want to do is this (pseudocode):

```
edges = part.edges()
edges[n].fillet(r)
```

Or ideally have the ability to do something like this (though I can't find any methods for edge properties). Pseudocode:

```
edges = part.edges()
for edge in edges:
    if edge.length() > x:
        edge.fillet(a)
    else:
        edge.fillet(b)
```

This would be very useful when a design contains non-orthogonal faces. I understand that I can select edges with selectors, but I find them unnecessarily complicated and work best with orthogonal faces. FreeCAD lets you treat edges as a list.

I believe there might be a method to select the closest edge to a point, but I can't seem to track it down.

If someone can provide guidance that would be great -- thank you!

Bonus question: Is there a way to return coordinates of geometry as a list or vector? e.g.:

```
origin = cq.workplane.center().val
>> [x,y,z]
```
(or something like the above)

jmwright, 2022-05-07:
You can get a list of edges by using `all()` and then grab a specific index or iterate over the edges manually, but there's no guarantee from the CAD kernel or from CadQuery that you will always get them in the same order.

```python
print(result.edges().all()[2])
```

```python
result = cq.Workplane().box(10, 10, 10)

for edge in result.edges().all():
    print(edge)
```

There is a [NearestToPoint](https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=nearest#cadquery.NearestToPointSelector) selector. See the [selector class reference](https://cadquery.readthedocs.io/en/latest/apireference.html#selectors) for many more that exist to accomplish different tasks.

TristanAntonsen, 2022-05-07:
Awesome, thank you very much! This makes a lot more sense with those extended selectors.

TristanAntonsen, 2022-05-07:
Awesome, thank you! Here's a simple test showing how I got it to work, for anyone else wondering:

```
import cadquery as cq

plane = cq.Workplane()
size = 8
radius = 1
point = (size * .375,size *.25, size * .95)

#test geometry
geometry = plane.rect(size,size).extrude(size).chamfer(size/4)

#visualizing point location
point_viewer = plane.sphere(.25).translate(point)

#selecting the edge
selector = cq.selectors.NearestToPointSelector(point)
selected_edges = geometry.edges(selector)

final_shape = selected_edges.fillet(radius)
```

[image]

michaelgale, 2022-05-11:
You can also use CQ-Kit (add-on utility library for CadQuery).  It can filter edges in a variety of ways including edge length:

```python
from cqkit import EdgeLengthSelector

# fillet edges length = 8
final_shape = geometry.edges(EdgeLengthSelector(8)).fillet(radius)

# fillet edges with length > 5
final_shape = geometry.edges(EdgeLengthSelector(">5")).fillet(radius)

# fillet edges with lengths 8 and 10
final_shape = geometry.edges(EdgeLengthSelector([8, 10])).fillet(radius)
```