CadQuery/cadquery #1090: extrude in subtractive mode ignores both parameter

lorenzncode, 2022-05-24:
```
import cadquery as cq

w, l, h = 50, 20, 15
lsnap = 8
rsnap = 0.5

s = cq.Sketch().circle(rsnap)

b = (
    cq.Workplane()
    .box(w, l, h)
    .tag("box")
    .faces("<Y")
    .vertices("<X and >Z")
    .workplane(centerOption="CenterOfMass", offset=-l/2)
    .center(0, -2.0)
    .placeSketch(s)
    .extrude(lsnap, combine="a", both=True)
    .center(0, -3.0)
    .placeSketch(s)
    # subtractive, both is ignored
    .extrude(lsnap, combine="s", both=True)
)
```

YHRen, 2023-01-13:
encountered the same problem. `both` in extrude is ignored after `placeSketch`.

YHRen, 2023-01-13:
when `combine="s"`, it defers to "cutBlind" which ignores "both".

closed 2024-07-13 by lorenzncode