CadQuery/cadquery #1092: Sweep circle in the normal direction

RemDelaporteMathurin, 2022-05-31:
```python
import cadquery as cq

pts = [
    (1, 1, 1),
    (1, 2, 2),
    (1, 3, 3),
]
p = cq.Workplane("XY").spline(pts, includeCurrent=True)
s = cq.Workplane("XY").circle(1).sweep(p)
```

Based on #296 , I would like to have the extrusion of the circle in the normal direction only (the idea is to make a classic tube from a set of points).

Unfortunately, this code produces a distorded tube:

[image]

Thanks for the help!

Jojain, 2022-05-31:
You need to have the normal of your circle aligned with the tangent of your spline at the start :
```py
pts = [
    (1, 1, 1),
    (1, 2, 2),
    (1, 3, 3),
]
p = cq.Workplane("XY").spline(pts, includeCurrent=True)
dire = p.val().tangentAt(0)
plane = cq.Plane((0,0,0), normal=dire)
s = cq.Workplane(plane).circle(1).sweep(p)
```

RemDelaporteMathurin, 2022-05-31:
Thanks for the hint!

closed 2022-05-31 by RemDelaporteMathurin

Jojain, 2022-05-31:
Note that this is true for any curve (even straight lines) not only splines

jmwright, 2022-05-31:
@RemDelaporteMathurin If what you're displaying is the exported STL, you may need to adjust your tolerance settings on the export call. Does it display fine in CQ-editor?

RemDelaporteMathurin, 2022-05-31:
> @RemDelaporteMathurin If what you're displaying is the exported STL, you may need to adjust your tolerance settings on the export call. Does it display fine in CQ-editor?

Yes this is the STL. I'll play with the tolerances then! Thanks! I didn't use the CQ-editor I'm afraid

jmwright, 2022-05-31:
You could export a STEP and use an online viewer for a double check too.