CadQuery/cadquery #1128: Cheatsheet/Quick Reference Creation

jpoles1, 2022-07-27:
Coming from OpenSCAD, I sometimes have trouble remembering the names and params of common commands in CadQuery. I had this exact same problem with OpenSCAD as overall my usage was intermittent. 

Seeing the most common commands and their params on one page would be incredibly helpful! I think it'd be great to have a page modeled after: https://openscad.org/cheatsheet/

I saw that there's a partial cheatsheet here: https://cadquery.readthedocs.io/en/latest/_static/cadquery_cheatsheet.html

I'd like to help out in creating such a resource, if it'd be helpful. Is there any reason not to go ahead and do so? Is the API fairly stable overall at this point?

jmwright, 2022-07-27:
@jpoles1 I think it would be fine if you want to create an updated cheat sheet. I created that CQ cheat sheet several years ago and have never really updated it because this issue is one of only two I can remember that ever referenced it, so I figured it wasn't being used. However, I don't think we ever linked to the cheat sheet in a prominent location, so that could be the reason more people don't use it. If you want to create one that's updated and more complete, we can link to it from the readme.

adam-urbanczyk, 2022-11-06:
Closed by #1129

closed 2022-11-06 by adam-urbanczyk