CadQuery/cadquery #1131: Centering Sketch.rect()

jpoles1, 2022-07-31:
I really like the ability to choose which axis to center on when drawing a Workplance.rect(). Is this something we can implement for Sketch.rect()? I figure it should be fairly straightforward to add a similar parameter to Sketch, but not sure.

adam-urbanczyk, 2022-07-31:
The idea behind `Sketch` was to keep the API relatively simple. Do you have an example that requires `Sketch` and justifies such an addition?

jpoles1, 2022-07-31:
To me this is more of a convenience function, it's something I use all the time for all sorts of designing tasks. It's not absolutely necessary but is both faster and in my opinion yields cleaner code.

For instance, I am currently cutting a hole in the wall of a box using a Sketch.rect(). In this case I wish to draw the hole with distances to other objects relative to the bottom edge of the hole. I could use Workplane.rect(..., centered=[1, 0]) which would center on horizontal but not the vertical axis. When using sketch however I have to draw the rect as well as translate the sketch over by half the hole's height in order to align the bottom edge with the feature of interest (see code below).

 ```
       hole_height = 14
       hole_width = 14
       orig = orig.faces("<Y").moveTo(0, floor_thick + hole_height/2)\
            .sketch().rect(hole_width, hole_height).vertices().fillet(2).finalize().cutThruAll()
```

Full file attached below (as .txt so github will let me upload).