CadQuery/cadquery #1134: Provide option to fuse faces with glue in Sketch

lorenzncode, 2022-08-21:
See https://github.com/CadQuery/cadquery/blob/d4cdeeb30d4549848d5dcd1561386a44af77e3f2/cadquery/sketch.py#L484

fuse is called without specifying glue (default is False).

Here is a case where providing an option to fuse with glue and given tolerance might be useful.

```python
s = (
    cq.Sketch()
    .parray(0, 0, 360, 4)
    .segment((0, 0), (3, 0))
    .segment(3, 45)
    .segment(4.2426, 135)
    .segment(3, 225)
    .close()
    .assemble()
    .clean()
)

print(f"Number of faces = {len(s.reset().faces()._selection)}")
```

The number of faces is 4.  clean() fails to combine to single face in this case.  

In other cases, clean() results in single face:

```python
s = (
    cq.Sketch()
    .push([(1,1), (-1,1)])
    .rect(2,2)
    .clean()
)
```

To not rely on clean() the faces could potentially be fused with glue.

```python
s = (
    cq.Sketch()
    .parray(0, 0, 360, 4)
    .segment((0, 0), (3, 0))
    .segment(3, 45)
    .segment(4.2426, 135)
    .segment(3, 225)
    .close()
    .assemble()
    .clean()
)

s._faces = s._faces.fuse(*s._faces.Faces(), glue=True, tol=1e-4)

print(f"Number of faces = {len(s.reset().faces()._selection)}")

r = cq.Workplane().placeSketch(s).extrude(3).faces(">Z")
print(f"Number of faces = {len(r.vals())}")
```

Potential solutions
1. Add glue, tol, parameters to `assemble`, and `each`?
   Calling `assemble` would provide an option to fuse with glue.

2. Provide a `Sketch` `combine` method to fuse faces with glue?

3. No code changes - user can manually fuse as above?

The parray with radius 0 example is based on a Discord discussion where @sethfischer is sketching the profile of an extrusion in a single "quadrant".

lorenzncode, 2022-08-21:
The specified precision of the segment length was too low:
` .segment(4.2426, 135)`

Maybe exposing glue and tol would be useful for cases where input data precision is low?

adam-urbanczyk, 2022-08-23:
I think you want `tol` (fuzzy bool ops) and not `glue` (AFAIR faster fusing of touching shapes). Yeah, exposing `tol` might be useful.