CadQuery/cadquery #1135: show_object(result) is not imported  using from jupyter_cadquery.viewer.client import show_object on ./CadQuery/cadquery/examples

x86girl, 2022-08-22:
There is at least this code line
```
from jupyter_cadquery.viewer.client import show_object
```
missing in some examples, like /Ex022_Revolution.py

jmwright, 2022-08-22:
Hi @x86girl 

jupyter-cadquery is not maintained by the core CadQuery team (although it is built on top of CadQuery), and so the examples are designed to be used with [CQ-editor](https://github.com/CadQuery/CQ-editor), which is supported by the core team. There is some cross-over (there is a .ipynb file in the examples directory), but we don't directly support jupyter-cadquery imports in the examples. However, this should probably be stated in the CadQuery readme, and/or a readme in the examples directory. Instructions on how to run the examples in jupyter-cadquery would also be a helpful addition since jupyter-cadquery is an important part of the ecosystem.

x86girl, 2022-08-22:
@jmwright I'm still understand how to setup things to develop 3d model using Jupyter and CQ-editor.
When I'm ready to, I will create a pull request with these info if you don't mind.
Thank you!

jmwright, 2022-08-22:
> When I'm ready to, I will create a pull request with these info if you don't mind.

@x86girl Sounds good, thanks.