CadQuery/cadquery #1137: ipython install add-on not working

puterd, 2022-08-24:
pip install cadquery[ipython]==2.2.0b0

Returns:
zsh: no matches found: cadquery[ipython]==2.2.0b0

on mac, pip updated

jmwright, 2022-08-24:
Are you running an M1 Mac?

puterd, 2022-08-25:
No, on pre M1/M2. I tried reinstalling ipython but still no luck.

jmwright, 2022-08-25:
What happens if you install cadquery by itself and then install ipython? Do you get the same error?

jmwright, 2022-08-27:
@puterd I'm going to leave this open until I have time to investigate why the install line given in the readme does not work.

fpq473, 2022-08-29:
The issue is with how zsh treats `[` and `]`.  This fails with the same message:
```sh
% echo cadquery[ipython]==2.2.0b0    
zsh: no matches found: cadquery[ipython]==2.2.0b0
% 
```
To fix, you can add quotes, e.g.
```sh
% pip install 'cadquery[ipython]==2.2.0b0'
...
```

closed 2022-08-30 by puterd