CadQuery/cadquery #1138: Sketch.located does not seem to have any effect

WizardUli, 2022-08-27:
Sketch.located does not seem to have any effect whenever and wherever it is called.
Perhaps an example of what it is supposed to do would help. I see that it is setting `self.locs` but it really does not seem to do anything whether I call it at the beginning/middle/end of sketching, it also does not seem to affect `Workplane.placeSketch` and I'm out of ides of what it should affect and when.

adam-urbanczyk, 2022-08-30:
Here is an example:

```python
from cadquery import *

s1 = Sketch().segment((0, 0), (1, 0)).segment((1, 1)).close().assemble()

w1 = (
    cq.Workplane()
    .add(s1)
    .extrude(1)
    )

w2 = (
    cq.Workplane()
    .add(s1.located(loc=Location((1,1,0))))
    .extrude(1)
    )
```

See also: https://github.com/CadQuery/cadquery/blob/d4cdeeb30d4549848d5dcd1561386a44af77e3f2/cadquery/sketch.py#L1007

WizardUli, 2022-08-30:
Thank you.
Now I also see the difference between `.placeSketch(sketch)` and `.add(sketch)`.