CadQuery/cadquery #1139: Please release #1083 (quality controls to STEP)

mojca, 2022-08-29:
I would like to request a release that includes #1083 since that prevents running the kicad scripts out-of-the-box

https://gitlab.com/kicad/libraries/kicad-packages3D-generator/-/merge_requests/24

adam-urbanczyk, 2022-08-29:
Can you provide more context? E.g. how do you install CQ and what prevents you from installing from the master?

mojca, 2022-08-29:
In short:
```
python -m pip install cadquery==2.2.0b0
```
I'm using Windows, Python 3.10 and FreeCAD 0.20 that comes with bindings precompiled for Python 3.8.

What prevented me from installing from the master is mainly realizing that the version from master is needed in the first place.

It took me hours to figure out how to run the scripts from https://gitlab.com/kicad/libraries/kicad-packages3D-generator/-/tree/master/LED_SMD, hitting tons of different errors on the way.

I first installed `pip install cadquery`. But then some of the scripts wanted to me to have FreeCAD installed as well (it wasn't visible from the system python). Then I installed cadquery via
```
"C:\Program Files\FreeCAD 0.20\bin\python.exe" -m pip install cadquery
```
and at some point it refused to work because cadquery was too old. Eventually (after tons of other failed attempts) I installed it with
```
"C:\Program Files\FreeCAD 0.20\bin\python.exe" -m pip install cadquery==2.2.0b0
```
but that still failed due to the missing functionality that has only been added recently.

I kept staring at https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Assembly.save and scratching my head since the documentation was listing the flag as available.

jmwright, 2022-08-29:
Hi @mojca , there are two things you can try.

First, install the new beta package from PyPi: `pip install cadquery==2.2.0b1`

Second, any time you need the latest of what is in the main branch if there is not a release (beta or otherwise): `pip install git+https://github.com/CadQuery/cadquery.git`

mojca, 2022-08-29:
> Hi @mojca , there are two things you can try.
> 
> First, install the new beta package from PyPi: `pip install cadquery==2.2.0b1`

Awesome, thank you very much. I tested it and the file generation seems smooth with this version.

(And in particular many thanks for making the kicad scripts independent of FreeCAD, so that I don't need to resort to "FreeCAD-specific" python.)

Now, https://pypi.org/project/cadquery/2.2.0b1/ still contains readme saying `pip install cadquery==2.2.0b0` and https://github.com/CadQuery/cadquery/tree/2.2.x doesn't contain the patch, but it's certainly going into direction where I would want it to go.

> Second, any time you need the latest of what is in the main branch if there is not a release (beta or otherwise): `pip install git+https://github.com/CadQuery/cadquery.git`

I did try that earlier with success, but only after "the war was over".

jmwright, 2022-08-29:
> still contains readme saying pip install cadquery==2.2.0b0

I've put updating the readme on my todo list, thanks.

mojca, 2022-08-31:
> See #1140 on a more universal pip command. Otherwise @mojca I think this issue is better to be discussed on the KiCAD tracker - not all CQ contributors are interested in KiCAD model generation scripts and whether they have properly documented list of requirements.

Thank you, the linked README change combined with the beta release provides the perfect fix combination so far, thank you very much.

I don't expect you to worry about or discuss any KiCad-related issues here (beyond making sure that the release of CQ supporting new featutes is intuitive enough for newbies to install). Thanks again.

closed 2022-08-31 by jmwright